maximkulkin/esp-homekit #135: Can’t pair with the flashed device

benchuk, 2020-04-14:
Hi,
I have build the firmeware using the latest esp idf docker env
Everything seems to build
Ok and I am able
To flash and monitor the app
But I cannot discover the homekit device on my ios device, seems to me like an mdns issue, but I am not sure
Cloud it be becuase
I am building over esp idf and not the rtos one?

Thanks

maximkulkin, 2020-04-14:
When trying LED example (I assume examples/esp32/led from esp-homekit-demo repository), did you set up wifi.h correctly? What do you see in logs on device?

benchuk, 2020-04-14:
Thanks for the replay,
I have updated the wifi file and I can verify that I am connected, I do have ping and the esp is listed on the router dhcp, so as for wifi connectivity everything seems to be ok

maximkulkin, 2020-04-14:
Can you show logs?

benchuk, 2020-04-14:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7060
load:0x40078000,len:14816
load:0x40080400,len:4456
entry 0x400806e8
I (72) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v4.1-dev-1572-g30372f5a4 2nd stage bootloader
I (39) boot: compile time 00:59:59
I (39) boot: Enabling RNG early entropy source...
I (45) boot: SPI Speed      : 40MHz
I (49) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00200000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x71034 (462900) map
I (276) esp_image: segment 1: paddr=0x0008105c vaddr=0x3ffb0000 size=0x05038 ( 20536) load
I (285) esp_image: segment 2: paddr=0x0008609c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /opt/esp/idf/components/freertos/xtensa_vectors.S:1778

I (286) esp_image: segment 3: paddr=0x000864a4 vaddr=0x40080400 size=0x09b74 ( 39796) load
I (309) esp_image: segment 4: paddr=0x00090020 vaddr=0x400d0020 size=0x13bd2c (1293612) map
0x400d0020: _stext at ??:?

I (772) esp_image: segment 5: paddr=0x001cbd54 vaddr=0x40089f74 size=0x131b8 ( 78264) load
0x40089f74: reconfigureAllWdts at /opt/esp/idf/components/esp32/panic.c:405

I (823) boot: Loaded app from partition at offset 0x10000
I (823) boot: Disabling RNG early entropy source...
I (823) psram: This chip is ESP32-D0WD
I (829) spiram: Found 64MBit SPI RAM device
I (832) spiram: SPI RAM mode: flash 40m sram 40m
I (838) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (845) cpu_start: Pro cpu up.
I (848) cpu_start: Application information:
I (853) cpu_start: Project name:     esp32-homekit-camera
I (859) cpu_start: App version:      2a26d79-dirty
I (865) cpu_start: Compile time:     Apr 14 2020 01:00:11
I (871) cpu_start: ELF file SHA256:  3475f48eeddb6cf4...
I (877) cpu_start: ESP-IDF:          v4.1-dev-1572-g30372f5a4
I (883) cpu_start: Starting app cpu, entry point is 0x4008157c
0x4008157c: call_start_cpu1 at /opt/esp/idf/components/esp32/cpu_start.c:276

I (0) cpu_start: App cpu up.
I (1775) spiram: SPI SRAM memory test OK
D (1775) memory_layout: Checking 8 reserved memory ranges:
D (1775) memory_layout: Reserved memory range 0x3f800000 - 0x3fc00000
D (1780) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (1787) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffbce48
D (1793) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (1800) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (1806) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (1813) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /opt/esp/idf/components/freertos/xtensa_vectors.S:1778

D (1819) memory_layout: Reserved memory range 0x40080000 - 0x4009d12c
0x40080000: _WindowOverflow4 at /opt/esp/idf/components/freertos/xtensa_vectors.S:1778

D (1826) memory_layout: Building list of available memory regions:
D (1832) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (1839) memory_layout: Available memory region 0x3ffbce48 - 0x3ffc0000
[... 356 lines cut ...]
I (6537) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (6547) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
D (6557) intr_alloc: Connected src 32 to int 17 (cpu 0)
D (6567) camera: Line width (for DMA): 5120 bytes
D (6567) camera: DMA buffer size: 2560, DMA buffers per line: 2
D (6577) camera: DMA buffer count: 8
D (6577) camera: DMA buffer total: 20480 bytes
D (6587) camera: Allocating DMA buffer #0, size=2560
D (6587) camera: Allocating DMA buffer #1, size=2560
D (6597) camera: Allocating DMA buffer #2, size=2560
D (6597) camera: Allocating DMA buffer #3, size=2560
D (6607) camera: Allocating DMA buffer #4, size=2560
D (6607) camera: Allocating DMA buffer #5, size=2560
D (6617) camera: Allocating DMA buffer #6, size=2560
D (6617) camera: Allocating DMA buffer #7, size=2560
I (6627) camera: Allocating 2 frame buffers (75 KB total)
I (6637) camera: Allocating 37 KB frame buffer in OnBoard RAM
I (6647) camera: Allocating 37 KB frame buffer in OnBoard RAM
D (6647) intr_alloc: Connected src 22 to int 18 (cpu 0)
D (6647) camera: Setting frame size to 640x480
>>> HomeKit: Starting server
I (6837) esp32_camera: Initializing streaming
>>> HomeKit: Using existing accessory ID: 04:DF:6E:D0:95:55
I (6857) esp32_camera: Streaming initialized
**>>> HomeKit: Configuring mDNS**

benchuk, 2020-04-16:
Ok after some investigations I have found the following:
1. this problem was found after building and installing with docker  (pre installed build env and esp idf version) using this docker image: [lpodkalicki/esp32-toolchain](https://hub.docker.com/r/lpodkalicki/esp32-toolchain)
2. after discovering this issue I have tried the **latest** version of docker by esp from docker hub [latest](https://hub.docker.com/layers/espressif/idf/release-v4.2/images/sha256-bcd8fc02d2345f01f842560b7246097495ab2caeebadeffef261d700239818f9?context=explore)
all version can be found here: [espressif/idf](https://hub.docker.com/r/espressif/idf/tags)
3. at the end the version of docker and idf that worked for me was version [3.3](https://hub.docker.com/layers/espressif/idf/release-v3.3/images/sha256-812064c2ff27c7d3c1f4d61b675a0c0a98bb4ab0a3fb4d42270db348f65bf939?context=explore)

not sure why but it works

maximkulkin, 2020-04-17:
When testing with esp-idf, I'm running latest master version and it works fine. Although I do not use Docker with esp-idf as it has pretty good scripts to install all dependencies on host machine.

closed 2020-06-26 by maximkulkin