maximkulkin/esp-homekit #138: Cannot combine GARAGE_DOOR_OPENER with other services

killobatt, 2020-04-16:
Hi. I'm trying to create a garage door opener using ESP32, which is two light zones, temp + humidity sensor, and I'm planning to add a CO sensor in future. 

However, I've found an issue with combining different services. 

Here is my configuration:

```c

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_other, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "garage"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "killobatt"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "237A2BAB119E"),
            HOMEKIT_CHARACTERISTIC(MODEL, "garage"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, led_identify),
            NULL
        }),
        HOMEKIT_SERVICE(GARAGE_DOOR_OPENER, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Garage opener"),
            HOMEKIT_CHARACTERISTIC(
                CURRENT_DOOR_STATE, HOMEKIT_CHARACTERISTIC_CURRENT_DOOR_STATE_CLOSED,
                .getter=gdo_current_state_get,
                .setter=NULL
            ),
            HOMEKIT_CHARACTERISTIC(
                TARGET_DOOR_STATE, HOMEKIT_CHARACTERISTIC_TARGET_DOOR_STATE_CLOSED,
                .getter=gdo_target_state_get,
                .setter=gdo_target_state_set
            ),
            HOMEKIT_CHARACTERISTIC(
                OBSTRUCTION_DETECTED, false,
                .getter=gdo_obstruction_get,
                .setter=NULL
            ),
            NULL
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary = false, .characteristics = (homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Front light"),
            HOMEKIT_CHARACTERISTIC(
                ON, false,
                .getter=frontlight_on_get,
                .setter=frontlight_on_set
            ),
            NULL
        }),
        // Why doesn't thermo services work among with the garage? Simulator works.
        HOMEKIT_SERVICE(TEMPERATURE_SENSOR, .primary = false, .characteristics = (homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Temperature sensor"),
            &temp_hum_sensor.temperature,
            NULL
        }),
        HOMEKIT_SERVICE(HUMIDITY_SENSOR, .primary = false, .characteristics = (homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Humidity sensor"),
            &temp_hum_sensor.humidity,
            NULL
        }),
        NULL
    }),
    NULL
};

...

temp_hum_sensor.temperature = HOMEKIT_CHARACTERISTIC_(CURRENT_TEMPERATURE, 0);
temp_hum_sensor.humidity = HOMEKIT_CHARACTERISTIC_(CURRENT_RELATIVE_HUMIDITY, 0);
```

When I'm commenting out temp/humidity sensor services, I can successfully pair with iOS Device (iOS 13.4.1).
[... 278 lines cut ...]
homekit_server_on_url url: /pairings
>>> homekit_server_on_pairings: HomeKit Pairings
>>> homekit_server_on_pairings: Free heap: 195600
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x04
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> tlv_debug: Type 1 value (36 bytes): 166DDF3F-ACD4-42A7-B075-085D1A678228
>>> HomeKit: [Client 55] Remove Pairing
>>> HomeKit: Removed pairing with 166DDF3F-ACD4-42A7-B075-085D1A678228
>>> HomeKit: Last admin pairing was removed, enabling pair setup
>>> HomeKit: Configuring mDNS
>>> homekit_setup_mdns: Accessory Setup ID = 2A5B
>>> send_tlv_response: [Client 55] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 55] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 3\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02
>>> homekit_client_process: [Client 55] Finished processing
>>> HomeKit: [Client 55] Closing client connection
TIMEOUT: 75E (80582) SENSOR: DHT22_ERROR_TIMEOUT
I (90582) SENSOR: Read temperature: 25.7, humidity: 52.3
I (100582) SENSOR: Read temperature: 25.6, humidity: 52.4
I (110582) SENSOR: Read temperature: 25.6, humidity: 52.7
```

I initially assumed I did something wrong in the config, but when I create the same accessory config in the Homekit Accessory Simulator, iPhone is able to pair and control the simulated accessory.

maximkulkin, 2020-04-17:
Ok, I have built accessory with your services and it worked fine.

Then I decoded your accessories JSON and found this problem with CURRENT_TEMPERATURE characteristic:
```json
            {
              "aid": 1,
              "iid": 18,
              "type": "00000011-0000-1000-8000-0026BB765291",
              "perms": [
                "pr",
                "ev"
              ],
              "ev": false,
              "description": "Current Temperature",
              "format": "float",
              "unit": "celsius",
              "minValue": 6.06114274642742e-39,
              "maxValue": 1.69625058101164e-36,
              "minStep": -1.62387431414683e-29,
              "value": 25.7000007629395
            }
```

Notice minValue, maxValue and minStep are all messed up (and thus current value is invalid). Same problem with CURRENT_HUMIDITY characteristic. You're probably initializing them in non-static context, and those characteristics need min/maxValue+minStep, which are defined as pointers (so you can opt in/out for those attributes), thus they are initialized with pointers to one-element arrays that are allocated on stack. Thus, when you copy data to proper structures, pointers remain pointers on stack and when you exit that subroutine they get garbage data.

To avoid that you should either initialize those characteristics in static context (e.g. global variable) thus those pointers will remain intact; or use NEW_HOMEKIT_CHARACTERISTIC() which will create A PROPER copy of characteristic (event if it is defined on stack).

killobatt, 2020-04-17:
That worked. Thanks!

closed 2020-04-17 by killobatt