maximkulkin/esp-homekit #139: The range does not match the actual environment

CHNleing, 2020-04-17:
Ask a question, the air conditioning temperature range: 16-30. I use HEATER_COOLER range is 0-100, 0-25. How to change this range. In line with air conditioning.

pbendersky, 2020-04-17:
When you create the characteristic, you can add a `min_value` and a `max_value`. Here's min, for a `HEATER_COOLER`:

```
            HOMEKIT_CHARACTERISTIC(
                COOLING_THRESHOLD_TEMPERATURE, 0,
                .getter=ac_cooling_threshold_get,
                .setter=ac_cooling_threshold_set,
                .min_value = (float[]) {16},
                .max_value = (float[]) {30},
                .min_step = (float[]) {1}
            ),
```

CHNleing, 2020-04-18:
@pbendersky 
Could you please take a look at this
I want to delete the automatic mode
`HOMEKIT_CHARACTERISTIC(
          TARGET_HEATER_COOLER_STATE, 0,
          .getter=air_target_state_get,
          .setter=air_target_state_set,
          .min_value = (float[]) {1},
          .max_value = (float[]) {3},
          .min_step = (float[]) {1},
          .valid_values = {
                    .count = 3,
                    .values = (uint8_t[]) { 1, 2, 3 }
                    }
          );`

CHNleing, 2020-04-18:
@pbendersky 
And how to get characteristics in real time.
After the cooling is set, the heating status is still displayed, and the cooling will be displayed only when you exit the home APP and reopen it.
Please advise, thank you

CHNleing, 2020-04-18:
When I use the remote control to turn off the air conditioner, I must quit the home app and then open the app to see the correct status

CHNleing, 2020-04-18:
solved

closed 2020-04-18 by CHNleing

peros550, 2020-04-19:
I have a working example of AC [here](https://github.com/peros550/esp-homekit-multiple-sensors/tree/master/examples/multiple_sensors_ac)

You may want to have a look if you like.