maximkulkin/esp-homekit #141: Clients not Disconnecting Properly?

mriksman, 2020-04-22:
How can I debug clients not disconnecting properly, and creating another client connection? It's filling up the maximum number of clients.

```
>>> HomeKit: Got new client connection: 49 from 192.168.0.38		
>>> HomeKit: Got new client connection: 50 from 192.168.0.41		
>>> HomeKit: [Client 49] Closing client connection		
>>> HomeKit: Got new client connection: 49 from 192.168.0.43		
>>> HomeKit: Got new client connection: 51 from 192.168.0.41		192.168.0.41
>>> HomeKit: [Client 50] Closing client connection		
>>> HomeKit: Got new client connection: 50 from 192.168.0.41		
>>> HomeKit: [Client 49] Closing client connection		
>>> HomeKit: Got new client connection: 49 from 192.168.0.40		192.168.0.40
>>> HomeKit: Got new client connection: 52 from 192.168.0.40		
>>> HomeKit: [Client 50] Closing client connection		
>>> HomeKit: Got new client connection: 50 from 192.168.0.38		
>>> HomeKit: Got new client connection: 53 from 192.168.0.41		192.168.0.41
>>> HomeKit: [Client 52] Closing client connection		
>>> HomeKit: Got new client connection: 52 from 192.168.0.41		
>>> HomeKit: [Client 50] Closing client connection		
>>> HomeKit: Got new client connection: 50 from 192.168.0.38		
>>> HomeKit: Got new client connection: 54 from 192.168.0.40		192.168.0.40
>>> HomeKit: [Client 52] Closing client connection		
>>> HomeKit: Got new client connection: 52 from 192.168.0.40		
>>> HomeKit: [Client 50] Closing client connection		
>>> HomeKit: Got new client connection: 50 from 192.168.0.41		192.168.0.41
>>> HomeKit: Got new client connection: 55 from 192.168.0.54		
>>> HomeKit: [Client 52] Closing client connection		
>>> HomeKit: Got new client connection: 52 from 192.168.0.41		
>>> HomeKit: Got new client connection: 56 from 192.168.0.54		
>>> HomeKit: [Client 55] Closing client connection		
>>> HomeKit: Got new client connection: 55 from 192.168.0.54		
>>> HomeKit: [Client 52] Closing client connection		
>>> HomeKit: Got new client connection: 52 from 192.168.0.41		192.168.0.41
>>> HomeKit: Got new client connection: 57 from 192.168.0.41		
>>> HomeKit: [Client 55] Closing client connection		
>>> HomeKit: Got new client connection: 55 from 192.168.0.54		
>>> HomeKit: Got new client connection: 58 from 192.168.0.41		192.168.0.41
>>> HomeKit: [Client 57] Closing client connection		
>>> HomeKit: Got new client connection: 57 from 192.168.0.41		
>>> HomeKit: [Client 55] Closing client connection		
>>> HomeKit: [Client 56] Closing client connection		
>>> HomeKit: Got new client connection: 55 from 192.168.0.54		
>>> HomeKit: Got new client connection: 56 from 192.168.0.41		192.168.0.41
>>> HomeKit: [Client 57] Closing client connection		
>>> HomeKit: Got new client connection: 57 from 192.168.0.41		
>>> HomeKit: Got new client connection: 59 from 192.168.0.38		
>>> HomeKit: Got new client connection: 60 from 192.168.0.41		192.168.0.41
>>> HomeKit: [Client 57] Closing client connection		
>>> HomeKit: Got new client connection: 57 from 192.168.0.41		
>>> HomeKit: [Client 59] Closing client connection		
>>> HomeKit: Got new client connection: 59 from 192.168.0.40		
>>> HomeKit: [Client 57] Closing client connection		
>>> HomeKit: [Client 59] Closing client connection		
>>> HomeKit: Got new client connection: 57 from 192.168.0.41		
>>> HomeKit: Got new client connection: 59 from 192.168.0.38		192.168.0.38
>>> HomeKit: [Client 57] Closing client connection		
>>> HomeKit: Got new client connection: 57 from 192.168.0.41		192.168.0.41
>>> HomeKit: [Client 55] Closing client connection		
>>> HomeKit: Got new client connection: 55 from 192.168.0.40		192.168.0.40
```

You can see `192.168.0.40` and `192.168.0.41` having multiple connections open.

What's the best way to manage this? Kick off duplicate connections?

mriksman, 2020-04-24:
I added ('borrowed' your code from another function) to function `void homekit_server_close_clients(homekit_server_t *server)` 
```
    char address_buffer[INET_ADDRSTRLEN];

    struct sockaddr_in addr;
    socklen_t addr_len = sizeof(addr);
    if (getpeername(context->socket, (struct sockaddr *)&addr, &addr_len) == 0) {
        inet_ntop(AF_INET, &addr.sin_addr, address_buffer, sizeof(address_buffer));
    } else {
        strcpy(address_buffer, "?.?.?.?");
    }

    CLIENT_INFO(context, "** %s **** homekit_server_close_clients ***** Discon %s", address_buffer, context->disconnect ? "Yes!" : "No");
```

And you can much easier see the offending list;
```
>>> HomeKit: No more room for client connections (max 12)
>>> HomeKit: [Client 58] ** 192.168.0.41 **** homekit_server_close_clients ***** Discon No
>>> HomeKit: [Client 59] ** 192.168.0.41 **** homekit_server_close_clients ***** Discon No
>>> HomeKit: [Client 60] ** 192.168.0.41 **** homekit_server_close_clients ***** Discon No
>>> HomeKit: [Client 57] ** 192.168.0.41 **** homekit_server_close_clients ***** Discon No
>>> HomeKit: [Client 56] ** 192.168.0.41 **** homekit_server_close_clients ***** Discon No
>>> HomeKit: [Client 54] ** 192.168.0.41 **** homekit_server_close_clients ***** Discon No
>>> HomeKit: [Client 55] ** 192.168.0.41 **** homekit_server_close_clients ***** Discon No
>>> HomeKit: [Client 53] ** 192.168.0.54 **** homekit_server_close_clients ***** Discon No
>>> HomeKit: [Client 52] ** 192.168.0.40 **** homekit_server_close_clients ***** Discon No
>>> HomeKit: [Client 51] ** 192.168.0.23 **** homekit_server_close_clients ***** Discon No
>>> HomeKit: [Client 50] ** 192.168.0.38 **** homekit_server_close_clients ***** Discon No
>>> HomeKit: [Client 49] ** 192.168.0.16 **** homekit_server_close_clients ***** Discon No
```

Any thoughts? Should HomeKit normally allow multiple connections? Maybe some code to kill older connections to the same IP?

HomeACcessoryKid, 2020-08-16:
The PR #165 I submitted recently and which has been merged solves this issue...

closed 2022-05-19 by mriksman