maximkulkin/esp-homekit #143: Removing accessories while ESP is off

vvrec, 2020-04-30:
If I remove the accessory in the phone application when the board is off, I cannot add it again until reupload the program.
Maybe you can add a full reset function from any interruption, or by clicking  "reset" on board?

peros550, 2020-04-30:
There is a reset function. Check out the repo esp-HomeKit-demo where Maxim has the examples and look for reset word in the code. You ‘ll see how reset is being called. 

On the other hand it would be nice if the firmware could identify situations like that in which user removed the accessory while accessory was shut down and reset its configuration itself.

vvrec, 2020-04-30:
homekit_storage_reset(); 

Sorry!  I was inattentive.  Thank!

closed 2020-04-30 by vvrec