maximkulkin/esp-homekit #145: With release 2.7.0 8826 board the phone is constantly losing accessory.

vvrec, 2020-05-06:
I went through all the possible sources of the problem until I found that with release 2.7.0 8826 board the phone is constantly losing accessory. 
Problem is gone when rolled back to the previous version

maximkulkin, 2020-05-06:
WAT? What is this version about?

vvrec, 2020-05-06:
Core ArduinoESP8266 for Arduino IDE https://github.com/esp8266/

maximkulkin, 2020-05-06:
esp-homekit does not support ArduinoESP8266 SDK. Supported SDKs are esp-open-rtos and ESP8266_RTOS_SDK from Espressif

vvrec, 2020-05-06:
I probably wrote incorrectly. I'm talking about an add-on for Arduino IDE to work with ESP8266 Board ...

maximkulkin, 2020-05-07:
Ok, I have two answers here: "It should work" and "It is not supported". Arduino IDE support usually means that it does provide you familiar functions (like digitalRead, digitalWrite, sleep, etc) which means it has an Arudino API layer. Even if it happen to compile, I suggest using supported SDKs directly.

mriksman, 2020-05-09:
I'm curious how you even got this going... Arduino IDE with ESP8266 uses NONOS - it shouldn't even run ESP-HomeKit. 
Are you using this; `https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266`? They ported it to support NONOS.

vvrec, 2020-05-15:
Issue is closed after updating to version 2.7.1.
For me ...

closed 2020-05-15 by vvrec