maximkulkin/esp-homekit #146: Combine led and temperature sensor

brisbaneeastcoast, 2020-05-11:
Hi,

I'm trying to combine the button project and the temperature_sensor project. 
I copied everthing from the build directory of the temperature_sensor project to the build folder of the button project. 
If I try to upload, the compiler says:
fatal error: No such dile or directory #include<button.h>

Do I have to update some build settings or something like that ?

Whould be great if someone can help me with this problem. 
[image]

maximkulkin, 2020-05-11:
Button example depends on [esp-button](https://github.com/maximkulkin/esp-button) library which you need to [depend upon in Makefile](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/button/Makefile#L5)

brisbaneeastcoast, 2020-05-11:
Thank you I just forgot the lines in the makefile.

closed 2020-05-11 by brisbaneeastcoast