maximkulkin/esp-homekit #149: Expand sysparam region space

beckmx, 2020-06-15:
Hello guys, a little doubt, not directly related to the library but as I am using the library I figured maybe someone has a clue.
 I would like to know how to expand the sysparam region space, I have an array of numbers that I would like to store, but it runs out of space because of the size of the array, I have 2 options:
1) increase the region of the sysparam ( I dont know how exactly), right now I have 4 sectors, I would like 7
2) Store my array directly as "set_data" but whenever I read the length it has much more indexes and is not the original sent

maximkulkin, 2020-06-15:
You can increase sysparam area size by setting a define in your Makefile:
```
CFLAGS += -DDEFAULT_SYSPARAM_SECTORS=14
```
(keep in mind that sysparams have two regions: active and standby, so you need twice as much sectors. 14 sectors are a total, so 7 sectors per region).

But I would not recommend using sysparams for that purpose at all. Seems like you're doing some kind of time series storage which deserves a dedicated area and probably a better optimized storage system. But of course the choice is yours.

beckmx, 2020-06-16:
wow, indeed it did the trick, I didn't use 14, because looks like my default was 4, so I increased to 6, and worked like a charm, I need ur PayPal link for those beers

closed 2020-06-16 by beckmx

maximkulkin, 2020-06-16:
Glad that it helped you.

If you want to support, I have Github Sponsors enabled (just open my profile page).