maximkulkin/esp-homekit #150: Adding multiple devices of same type fails

haeckse, 2020-06-16:
When I run homekit with a couple of different nodes, like a door sensor, a sonoff-switch, and a temperature sensor, everything works fine. As soon as I add another sensor of the same type, like a second door-sensor or temperature-sensor, the one that has been added before immediately shows "no response" then.

Also, when I have some unpaired sensors showing in the 'add device' screen, and I power another device, of which the same type exits already, both devices disappear in the list.

For example the monitor output of two temperature sensors having that issue:

```
"$pp_task_hdl : 3fff00e0, prio:14, stack:512
pm_task_hdl : 3ffef9d0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4050, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jun 14 2020 17:58:19
phy ver: 273, pp ver: 8.3

mode : sta(80:7d:3a:40:f3:9e)
add if0
HomeKit: Starting server
HomeKit: Formatting HomeKit storage at 0x100000
HomeKit: Generated new accessory ID: 76:F0:D0:E9:29:C9
HomeKit: Generated new accessory key
HomeKit: Configuring mDNS
mDNS announcement: Name=Temperatur Sensor
md=TF01pv=1.0id=76:F0:D0:E9:29:C9c#=1s#=1ff=0sf=1ci=9 Port=5556 TTL=4500
scandone
add 0
aid 1
cnt

connected with haecksenwerk, channel 4
dhcp client start...
ip:192.168.1.106,mask:255.255.255.0,gw:192.168.1.1

###

b$pp_task_hdl : 3fff00e0, prio:14, stack:512
pm_task_hdl : 3ffef9d0, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4050, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Jun 14 2020 17:58:19
phy ver: 273, pp ver: 8.3

mode : sta(2c:3a:e8:0a:61:ae)
add if0
HomeKit: Starting server
HomeKit: Formatting HomeKit storage at 0x100000
HomeKit: Generated new accessory ID: 1B:88:4F:A6:05:40
HomeKit: Generated new accessory key
HomeKit: Configuring mDNS
mDNS announcement: Name=Temperatur Sensor
md=TF01pv=1.0id=1B:88:4F:A6:05:40c#=1s#=1ff=0sf=1ci=9 Port=5556 TTL=4500
scandone
add 0
aid 1
cnt
connected with haecksenwerk, channel 4
dhcp client start...
ip:192.168.1.112,mask:255.255.255.0,gw:192.168.1.1

```

So, apart from the name, the mDNS announcements differ, and I even assigned different serials in the HOMEKIT_CHARACTERISTIC settings.

Any ideas if I missed setting up something in the configuration or anything else?

perdidor, 2020-06-18:
> mDNS announcement: Name=Temperatur Sensor

You need unique names as they are DNS names. Cheers.

haeckse, 2020-06-19:
@perdidor That indeed solves the issue. Thanks a lot! \o/

closed 2020-06-19 by haeckse