maximkulkin/esp-homekit #151: Some thoughts about "Accessory not responding after random amount of perfect runtime" problem

perdidor, 2020-06-18:
Hi all.
I've spent a lot of my lifetime diving deeper and deeper,  trying to understand the reasons of subj.
And, finally, here is my case:
DO NOT IMPLEMENT any thread-locking loops unless you are using RTOS async tasks. In my case i put infinite loop to the end of main(), When exception raised (bugs as usual) while this loop, it can crash whole homekit part, but http will be still available and this is fucking awesome behavior i think.
Solution:
if you need loop, put it someway like

```
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>

static void loop_task(void* arg)
{
	for(;;) {
		//TODO: loop content
	}
}
```

and create from main():

`xTaskCreate(loop_task, "loop_task", 2048, NULL, 10, NULL);`

of course, you can create as many tasks as you need. Cheers.

maximkulkin, 2020-06-19:
Do you mean that people that have random “not responding” problems have implemented firmware incorrectly, so it deadlocks or do you mean there is a problem with esp-homekit framework?

perdidor, 2020-06-19:
There are no new bugs investigated in esp-homekit framework (in my case), so it's just incorrect using RTOS features. It was hard to investigate and understand why accessory not responding while HTTP, mDNS and other things are alive. Ii saw relevant topics in issues and just sharing my experience. Hope this will help anybody.

maximkulkin, 2020-06-19:
Thank you for information, although I think people in other issues often use stock firmwares which do not run main loop inside user_init and still have those issues.

Thanks anyways

closed 2020-06-19 by maximkulkin