maximkulkin/esp-homekit #153: cant detect devices on iphone

lukasa1993, 2020-06-21:
```
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: FA:39:08:A5:D6:69
>>> HomeKit: Configuring mDNS
scandone
mDNS announcement: Name=Contact Sensor md=DS1pv=1.0id=FA:39:08:A5:D6:69c#=1s#=1ff=0sf=1ci=10 Port=5556 TTL=4500
```

this run properly no errors it took local ip, local ip is pingable from laptop but iphone still cant find devices to add in homekit i used default `.password` trying to build door sensor…

mriksman, 2020-06-22:
It's already got pairing information? `Using existing accessory ID: FA:39:08:A5:D6:69` 
Did you try and fail?
Use function `homekit_server_reset()` to erase the pairing data and try again.

lukasa1993, 2020-06-22:
```
>>> HomeKit: Starting server
>>> HomeKit: Resetting HomeKit storage
>>> HomeKit: Formatting HomeKit storage at 0x100000
scandone
>>> HomeKit: Generated new accessory ID: D0:15:78:42:5F:E2
add 0
aid 3
cnt

>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Contact Sensor md=DS1pv=1.0id=D0:15:78:42:5F:E2c#=1s#=1ff=0sf=1ci=10 Port=5556 TTL=4500
ip:192.168.1.106,mask:255.255.255.0,gw:192.168.1.1

```
iphone still cant find it

```
	homekit_server_reset();
    homekit_server_init(&config);

    homekit_characteristic_notify(&door_open_characteristic, door_state_getter());
```

i added that like that…

 by the way i flash with `erase_flash` in front of it shouldn't it reset any data it has on it ?

mriksman, 2020-06-22:
> by the way i flash with `erase_flash` in front of it shouldn't it reset any data it has on it ?

Yes. You should see 
`>>> HomeKit: Generated new accessory ID: D0:15:78:42:5F:E2`
Every time you load after `erase_flash` I would imagine...?

Try commenting out the line `homekit_characteristic_notify(&door_open_characteristic, door_state_getter());`. Any difference?

Have you tried one of the examples in `esp-homekit-demo`.

Post the rest of your code.

maximkulkin, 2020-06-22:
You do not need to reset device every time to generate a new accessory ID. While it surely means that the device is not paired with anything, it is sufficient if it does not display following line:

```
Found admin pairing with XX:XX:XX:XX:XX:XX, disabling pair setup
```

So, with your original log it should be fine for pairing, it just says that it has already generate ID on one of previous startups. It does not say that it has a pairing and pairing will be disabled.
```
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: FA:39:08:A5:D6:69
>>> HomeKit: Configuring mDNS
```
One other thing to notice is "sf" value in mDNS announcement: if it is "1" ("sf=1") then pairing is enabled, if it is "0" ("sf=0"), pairing is disabled.

@lukasa1993 Make sure that both iPhone and your accessory are in the same network.

lukasa1993, 2020-06-22:
@maximkulkin they are i can ping it from iphone

lukasa1993, 2020-06-22:
> Yes. You should see
> `>>> HomeKit: Generated new accessory ID: D0:15:78:42:5F:E2`
> Every time you load after `erase_flash` I would imagine...?
>
> Try commenting out the line [...]

i tried sonoff_basic and button as well also led_status all of them same doesn't come up in iphone

lukasa1993, 2020-06-22:
```
ESP-Open-SDK ver: 0.0.1 compiled @ Jun 21 2020 19:55:09
phy ver: 273, pp ver: 8.3

Using Sensor at GPIO4.
>>> HomeKit: Formatting HomeKit storage at 0x100000
Door state was requested (open).
mode : sta(fc:f5:c4:97:21:00)
add if0
>>> HomeKit: Starting server
>>> HomeKit: Resetting HomeKit storage
>>> HomeKit: Formatting HomeKit storage at 0x100000
scandone
>>> HomeKit: Generated new accessory ID: C9:83:73:C3:8E:58
add 0
aid 3
cnt

connected with test1234, channel 1
dhcp client start...
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Contact Sensor md=DS1pv=1.0id=C9:83:73:C3:8E:58c#=1s#=1ff=0sf=1ci=10 Port=5556 TTL=4500
ip:192.168.1.106,mask:255.255.255.0,gw:192.168.1.1
```

latest log sf=1 and all but still doesn't come up in iphone

lukasa1993, 2020-06-22:
something up with mdns i think when i scan network on that ip mdns is not detected… otherwise everything seems fine 

```
>>> HomeKit: Got new client connection: 4 from 192.168.1.103
>>> HomeKit: [Client 4] Closing client connection
```

i triggered this by manually telneting on ip and port. its just mdns i think

maximkulkin, 2020-06-22:
@lukasa1993 What kind of WiFi access point do you use? Can you try pinging it with it's local name (`Contact Sensor.local`)? Can you see it with mDNS browser (on macOS the command to do that is `dns-sd -B _hap._tcp local`)

lukasa1993, 2020-06-22:
`dns-sd -B _hap._tcp local` when i run this other homekit devices come up but not this one…

lukasa1993, 2020-06-22:
god dam it, it was connecting to IPS wifi dont even know how it was reaching it, on same network but probably blocks mdns… GGS guys i am all set

closed 2020-06-22 by lukasa1993