maximkulkin/esp-homekit #157: Force update when internal variable changes

pastre, 2020-07-02:
I have a working led example running on an esp32, and I'm using another sensor to update the `led_on` bool, but HomeKit is not updating on my cell when I change this variable. I know the variable is changing because when I modify `led_on` and restart the Home app on my phone the LED state changes. Do you have any idea if I can force the update?

maximkulkin, 2020-07-02:
To notify controllers that some characteristic value has changed, you need to call `homekit_characteristic_notify(characteristic, value);`.

E.g. check [this example](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/sonoff_basic/main.c#L109-L111). First, it updates value of characteristic, so if controller connects to accessory and requests all accessory characteristics, it will get proper value. But controllers that are already connected have no idea that something has changed. And that's why homekit_characteristic_notify() is there.

maximkulkin, 2020-07-02:
Setting characteristic value and sending notification is intentionally separate, so you can have finer control over those. E.g. with STATELESS_PROGRAMMABLE_SWITCH service, PROGRAMMABLE_SWITCH_EVENT characteristic has always null value, but you can [send notifications with actual values, without changing stored value](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/button/main.c#L44).

pastre, 2020-07-03:
Thanks! That worked :)

closed 2020-07-03 by pastre

ZweiEuro, 2021-02-19:
@maximkulkin How does the server internally differentiate between the different accessories ?
In your example they all use the same callback or event, with just their internal value changed. But where does it say which device is updated? Or is this handled with the callback ?

maximkulkin, 2021-02-19:
Well, a) you can match accessory by characteristic (since characteristic can belong to only one service of one accessory); b) you can store pointer to arbitrary data in characteristic's context field and use it to quickly match which accessory it is.

ZweiEuro, 2021-02-19:
I did it via a mix, the order in which the accessories doesn't really change so i just store the characteristic pointers and with the index i can instantly go to the one i need. But thanks ! So i just need the characteristic pointer

ZweiEuro, 2021-02-21:
@maximkulkin something else that came up, does this cause the callback to be called from the homekit_characteristic_notify?

maximkulkin, 2021-02-21:
Yes, characteristic callbacks will be called if you call homekit_characteristic_notify(). To avoid loops you can check if the value is the same as current and skip doing anything. Or you can have a global flag that you can set before calling homekit_characteristic_notify() and check it inside notification callback. Or use setter instead - that one is not called when you do homekit_characteristic_notify() (but make sure you update characteristic value if you go that route).

ZweiEuro, 2021-02-21:
My bad fix is settings is_null in the value and then the callback just ignores it and sets is_null back to false.
it works but it's not really clean.
just a manual circle break

maximkulkin, 2021-02-21:
Setting is_null will send "null" value notification to all controllers. That's not what you want.