maximkulkin/esp-homekit #158: Soft WDT reset

antonioreyna, 2020-07-10:
Hello now im getting this problem and it restarts.

the stack trace says.

```
0x4023c698: _vfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 531
0x4023bf06: vprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vprintf.c line 40
```

```
>>> [  67199] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c

Soft WDT reset

>>>stack>>>

ctx: cont
```

maximkulkin, 2020-07-10:
Code please

antonioreyna, 2020-07-10:
```
void loop() {
    // -- doLoop should be called as frequently as possible.
    iotWebConf.doLoop();
    arduino_homekit_loop();

    // Only needed in forced mode! In normal mode, you can remove the next line.
    bme.takeForcedMeasurement(); // has no effect in normal mode

    if (iotWebConf.getState() == IOTWEBCONF_STATE_ONLINE && strlen(serialNumberValue) != 0) {
        if (delay_check(HomeKit.last, HomeKit.refresh)) { // update battery voltage if timer has elapsed
            my_homekit_report();
            HomeKit.last = millis(); // update structure variable to current time
        }
    }

    delay(10);
}

extern "C" homekit_server_config_t config;
extern "C" homekit_characteristic_t cha_current_temperature;
extern "C" homekit_characteristic_t cha_humidity;
extern "C" homekit_characteristic_t cha_status_active;

static uint32_t next_heap_millis = 0;
static uint32_t next_report_millis = 0;

void my_homekit_setup() {
    arduino_homekit_setup(&config);
}

void my_homekit_report() {
    //Serial.println("Inicia report");
    cha_current_temperature.value.float_value = temp;
    cha_humidity.value.float_value = hum;
    cha_status_active.value.bool_value = true;
    //LOG_D("Current temperature: %.1f", temperature_value);
    //Serial.print("Current temperature: %.1f");
    //Serial.println(temp);
    homekit_characteristic_notify(&cha_current_temperature, cha_current_temperature.value);
    homekit_characteristic_notify(&cha_humidity, cha_humidity.value);
    homekit_characteristic_notify(&cha_status_active, cha_status_active.value);
    //Serial.println("Termina report");
}

```

maximkulkin, 2020-07-10:
That looks like ArduinoCore something, I only do support for esp-homekit, which uses SDK(s) based on FreeRTOS.

antonioreyna, 2020-07-12:
yes im using arduino to code the esp8266

mriksman, 2020-07-12:
Probably this https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266

closed 2020-07-13 by antonioreyna