maximkulkin/esp-homekit #159: QR code pairing

haeckse, 2020-07-15:
I have set up a device in access point mode with

`sdk_wifi_set_opmode(STATIONAP_MODE);`

The device shows up in the network then, and with the generated QR code I can start the pairing, but after a while the HomeKit app abandons with 'could not find device'.

Does the QR pairing only work when the device has been already added to the network with  `sdk_wifi_station_set_config(&wifi_config)`  respectively by doing it manually via the  `wifi_config_init(..)` function?

maximkulkin, 2020-07-15:
Not sure about commercial devices that can hide and automate process of configuring WiFi network on accessory, esp-homekit requires accessory to already be in WiFi network for pairing to work. Thus, either by hardcoding or via esp-wifi-config, but you need to configure WiFi first, then attempt QR pairing.

haeckse, 2020-07-15:
Thanks for clarifying.

I performed a reset on a commercial device, which is a lightbulb from Koogeek, to see how it presents itself in the network. It doesn't show up in the regular network-list, like an accessory with esp-wifi-config does, for example, but in a "Set up new device" list instead. There, I remember having seen popping up an Airplay printer lately. But there's no need to add the lamp to the network in the settings. Once one has scanned its code with the HomeKit app, the lamp starts pairing and one gets prompted for giving permission to add it to the network

In Wireshark, I can see the lamp sending frequent mDNS broadcasts with its MAC address.

[image]

[image]

closed 2020-07-15 by haeckse

AramVartanyan, 2020-07-15:
For this feature you will need MFi commercial license and also the Apple’s WAC (Wireless Accessory Configuration) library.

haeckse, 2020-07-15:
I see. I just took a look into Apple's MFi-FAQ's:

"The MFi Manufacturing License is intended solely for companies that own one or more manufacturing facilities."

Too bad :)