maximkulkin/esp-homekit #162: Unable to pair after latest updates

peros550, 2020-08-05:
I'm building a rather overloaded accessory. It includes motion, light , temp & humidity sensor and a heater which uses esp-ir to control a wall AC. Code is [here](https://github.com/peros550/esp-homekit-multiple-sensors)

After latest updates in esp-homekit I'm unable to pair new accessories. 

I have captured some logs 

```
SDK version:0.9.9
>>> wifi_config: Initializing WiFi config
!!! wifi_config: No configuration found
>>> wifi_config: Starting AP mode
>>> wifi_config: Starting DHCP server
mode : sta(38:2b:78:03:79:83) + softAP(3a:2b:78:03:79:83)
add if0
add if1
bcn 100
>>> wifi_config: Starting WiFi scan
>>> wifi_config: Starting DNS server
>>> wifi_config: Starting HTTP server
scandone
scandone
add 1
aid 1
station: 8c:86:1e:0e:16:d6 join, AID = 1
scandone
scandone
scandone
scandone
>>> wifi_config: Connecting to test0
>>> wifi_config: WiFi Mode: Normal
>>> wifi_config: Connecting to test0
>>> wifi_config: WiFi Mode: Normal
scandone
scandone
scandone
reconnect
scandone
reconnect
scandone
station: 8c:86:1e:0e:16:d6 join, AID = 1
>>> wifi_config: Connecting to test0
>>> wifi_config: WiFi Mode: Normal
scandone
scandone
reconnect
scandone
add 0
aid 3
cnt

connected with test0, channel 10
dhcp client start...
ip:192.168.177.36,mask:255.255.255.0,gw:192.168.177.1
>>> wifi_config: Connected to WiFi network
station: 8c:86:1e:0e:16:d6 leave, AID = 1
rm match
bcn 0
del if1
mode : sta(38:2b:78:03:79:83)
Function called without core lock
CONNECTED TO >>> WIFI <<<
HOSTNAME set>>>>>:HomekitSensor-037983
>>> HomeKit: Starting server
>>> HomeKit: Formatting HomeKit storage at 0x8c000
>>> HomeKit: Generated new accessory ID: CE:3C:51:31:D2:A6
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
>>> wifi_config: Stopping HTTP server
Timer Stop Failed
Timer Start Failed
[... 194 lines cut ...]
>>> homekit_server_on_pair_setup: Free heap: 22836
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 22752
>>> homekit_server_on_pair_setup: [Client 4] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 22892
>>> homekit_server_on_pair_setup: [Client 4] Using user-specified password: 333-33-333
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Failed to set SRP password (code -173)
>>> crypto_srp_get_public_key: Calculating public key
>>> send_tlv_response: [Client 4] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): \x09P1\x91\x1B\xB7\xFC\x07JI\x01f\x19\xAEv\\\x82\xFAS\x8[\x96\x1D]\x80\xE5\xC1Q'\xD0\xB7l\x13\xA9\x16\xD2/\x1F\x05\xEE\x9135\xAC\xCF\xA2=\xEB[&\xAFq~-~\xE6\xCC\x11\xA1\x86\xD2\xBA\x09\xA3I\x14:t\xD1O\x09+\xB5f\x99S\xD3<\x8C\xDE=\x0A\xE8\x17\xBF\xFDYxs\x9FP\xBDO\x9F\x18O\x8C\x08\xB7(<\xD2\x92\x1B\xF5>\xCB/\xD0\x12\xC3\xB0\x15\xEETI\x08\xA3\xEF\x8D\xF1&\x06\xC4l\x01\xCD`\x19Pm\xD9\xE3\xC13\xF9\xAE!!\x0Cjm\xF1\xF4%\xA6g\xFA\xC8py\x10\xD2nw\xDA^\xCB\x1EY\xCC\xFF\xCAD\x1C\xE0\x1B\xD8\xC8\x9Eq\xF4*\x97\x85\xF9\x11\x01s\xE1\x0C\x10x\x05~\xB8,nR\x9C\x8C.$\xE3\xF2p\x92q#5\xBF\xCB\x80\x18WL\x92\xA6\x87BLU\xD8*\x99\x00^\xB3\xA9\x88\xE6f\xD8\x115\xA1lJ\x15\xDF\x8B \xE2}\xEF=\x8B\x93xG6\xFBK\x119\xB9\x13\xD2F\xB6\xB0\xEE\xD7\x07\x13\x13\x9B\xDC*\xF3R\xB3\xC3\xB2\xDE\xCA\xB1\x96\xEB\xE5Yf\x06-\x14,\xC3,\xCB\x09\xC1Y\xB2\xA8[\x8EMR\xD6\x1E\x87V<<\x13\x90\x18(#\xC9\x86K\xCD\xD9\xF0\xFD\xB9\xEC\xD6\xE1\xD8K\xF6Fd\xB2\x1C7C^\xDD\xC7\xAB\xDB\xDCtkY\xC8y\xC8l~\xFD\xE9\x91\xBCM\xFC\xF2\xA6m\xF1\xE8'\xC7\xD0Z\xE6\x97D\xF8t:pV\xFF\xF6\xF1p]\x07\xCA\xB1\x0F\xCAI\xBE7}\xAE\xCA.%\x9Bnd\xE2\x0C\x95\xC3
>>> tlv_debug: Type 2 value (16 bytes): \xAA\x0B{]\xE4\xD2\xB2\x16!\xFDAmR2=\x13
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 4] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFF\x09P1\x91\x1B\xB7\xFC\x07JI\x01f\x19\xAEv\\\x82\xFAS\x8[\x96\x1D]\x80\xE5\xC1Q'\xD0\xB7l\x13\xA9\x16\xD2/\x1F\x05\xEE\x9135\xAC\xCF\xA2=\xEB[&\xAFq~-~\xE6\xCC\x11\xA1\x86\xD2\xBA\x09\xA3I\x14:t\xD1O\x09+\xB5f\x99S\xD3<\x8C\xDE=\x0A\xE8\x17\xBF\xFDYxs\x9FP\xBDO\x9F\x18O\x8C\x08\xB7(<\xD2\x92\x1B\xF5>\xCB/\xD0\x12\xC3\xB0\x15\xEETI\x08\xA3\xEF\x8D\xF1&\x06\xC4l\x01\xCD`\x19Pm\xD9\xE3\xC13\xF9\xAE!!\x0Cjm\xF1\xF4%\xA6g\xFA\xC8py\x10\xD2nw\xDA^\xCB\x1EY\xCC\xFF\xCAD\x1C\xE0\x1B\xD8\xC8\x9Eq\xF4*\x97\x85\xF9\x11\x01s\xE1\x0C\x10x\x05~\xB8,nR\x9C\x8C.$\xE3\xF2p\x92q#5\xBF\xCB\x80\x18WL\x92\xA6\x87BLU\xD8*\x99\x00^\xB3\xA9\x88\xE6f\xD8\x115\xA1lJ\x15\xDF\x8B \xE2}\xEF=\x8B\x93xG6\xFBK\x119\xB9\x13\xD2F\xB6\xB0\xEE\x03\x81\xD7\x07\x13\x13\x9B\xDC*\xF3R\xB3\xC3\xB2\xDE\xCA\xB1\x96\xEB\xE5Yf\x06-\x14,\xC3,\xCB\x09\xC1Y\xB2\xA8[\x8EMR\xD6\x1E\x87V<<\x13\x90\x18(#\xC9\x86K\xCD\xD9\xF0\xFD\xB9\xEC\xD6\xE1\xD8K\xF6Fd\xB2\x1C7C^\xDD\xC7\xAB\xDB\xDCtkY\xC8y\xC8l~\xFD\xE9\x91\xBCM\xFC\xF2\xA6m\xF1\xE8'\xC7\xD0Z\xE6\x97D\xF8t:pV\xFF\xF6\xF1p]\x07\xCA\xB1\x0F\xCAI\xBE7}\xAE\xCA.%\x9Bnd\xE2\x0C\x95\xC3\x02\x10\xAA\x0B{]\xE4\xD2\xB2\x16!\xFDAmR2=\x13\x06\x01\x02
>>> homekit_client_process: [Client 4] Finished processing
>>> HomeKit: [Client 4] Closing client connection
CLIENT JUST DISCONNECTED
```

peros550, 2020-08-05:
After more attempts, it was only possible to pair again by taking the library back to previous commit (9th of June 2020)
Let me know if you would like me to test a different branch.

maximkulkin, 2020-08-06:
I've checked your firmware (multiple_sensors_ac) and it pairs fine for me. Right from the start it has 27K to spare which is sufficient for pairing process.

Here is my log for comparison:
```
Opening /dev/tty.usbserial-14310 at 115200bps...
[b?g???o'???o'???cl
                   ?'o??|???n??clpp_task_hdl : 3fff2ae0, prio:14, stack:512
pm_task_hdl : 3fff2378, prio:1, stack:176
frc2_timer_task_hdl:0x3fff6eb0, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Aug  6 2020 00:57:45
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
>>> wifi_config: Initializing WiFi config
>>> wifi_config: Connecting to TheNetwork
mode : sta(84:0d:8e:ab:9d:41)
add if0
scandone
add 0
aid 11
cnt

connected with TheNetwork, channel 8
dhcp client start...
ip:10.0.1.40,mask:255.255.255.0,gw:10.0.1.1
>>> wifi_config: Connected to WiFi network
CONNECTED TO >>> WIFI <<<
HOSTNAME set>>>>>:HomekitSensor-AB9D41
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: AC:C7:7A:12:9F:25
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=HomekitSensor-AB9D41
md=esp8266pv=1.0id=AC:C7:7A:12:9F:25c#=1s#=1ff=0sf=1ci=9 Port=5556 TTL=4500
SERVER JUST INITIALIZED
>>> homekit_run_server: Staring HTTP server
ip:10.0.1.40,mask:255.255.255.0,gw:10.0.1.1
>>> HomeKit: Got new client connection: 4 from 10.0.1.11
CLIENT JUST CONNECTED
>>> homekit_client_process: [Client 4] Got 138 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 27692
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 27604
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 4] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 26940
>>> homekit_server_on_pair_setup: [Client 4] Using user-specified password: 222-22-222
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> send_tlv_response: [Client 4] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): \xFD\xD2\x89\xDD>\xB0\xED\x10\xD7;:-2\xFF\x9E"\xC8%k\x0FtEc\x1C\xACeK\xB8\xCC\xEB\xB6\xAANk\x08K#\x8C\xC0\xF3\xBE4\x18\x14\x89\xA1_){X\xF7^w\xA1/\xA4T\xA60a\x85D\x06i$\x14Ge&\xA5\xB7X;\xF3\xBD\x9E\xB1a\xE49mh\x91a\x81Q,#\xC4\x19\x0D0\xE7\x0D\x14}\xEF-\xF8\xC2\xD4\xCFy\x89\xB1\xB7i\xE8\x16`\xABq\xEB\xF2\xF3\xB4}\xB85\xAD\x08\xFF\x0C>!]\x9C\x01Iu\xD3A8\xFB\x8E\xB8\x16B\x0C\xB0!\xF7\x95\xBF/4\xAA\x89\xC8\xCF\x12\x12%G\xBC\xFB\xA9-\x1C\xE9\x1C^\xD5\xCA\x0E\xE7\xC7\xC6o\xD4\xD5\x89\xDA\x88\xC7p\x0EC\xCA\xE3\x18\x93-z\xA3\x01\x82\xCD\xA9\xF8\x0F\x86.1\xEA,\x05q\x0D\x0D-T\x91E\\\xD2r\xBD\x93\xE1!.>\x1C\xD1\xAFt^\xB1j\x99\xCF\xDB\x01\\\xF4\xD6h\x0E]\xA5\x8CR[\xFC5\xB1\xF9\xCA_\x14MmG\xA9W\xFD\x9F\xD2\x1A\x10c@\xD4w\x0Fr\xC3\x19\xF5\x15\xFB^\xDB\x15\xAE\xD3\xBE\xD8\xDB\xFB\xA8!=6\xB1\xE5\x92(\xAD\xA1\xFA\xB6;=\x03f\xD1!\x07S\x8A\xE1>2-\x0F-\xCC0C\x8B\xA6\x18J,$ \xB6\x0At\x13\x85\x15\xA6\xA48\x0A_IT\xD5\x09T\xA2`\xAD(\xAE"}\xFE\xCD\xD1\x9Ed\x08\xE4\x10\x00\xA4\xDF\x1D'\x08\xC2\x0F%>\x08\x9F\xFB\x96GZT\x89\x818&\xC6g5B\x86\x80\x0B\xB9\x98<\xD6\x9D\x9E\xC6(\x13oZ\xA5\xEC\xDC\x0E\xBEq
>>> tlv_debug: Type 2 value (16 bytes): $\xC5\xDB\xAAd\x80N\x9ES\xDDp\xFF\x8E\xE3#_
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 4] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFF\xFD\xD2\x89\xDD>\xB0\xED\x10\xD7;:-2\xFF\x9E"\xC8%k\x0FtEc\x1C\xACeK\xB8\xCC\xEB\xB6\xAANk\x08K#\x8C\xC0\xF3\xBE4\x18\x14\x89\xA1_){X\xF7^w\xA1/\xA4T\xA60a\x85D\x06i$\x14Ge&\xA5\xB7X;\xF3\xBD\x9E\xB1a\xE49mh\x91a\x81Q,#\xC4\x19\x0D0\xE7\x0D\x14}\xEF-\xF8\xC2\xD4\xCFy\x89\xB1\xB7i\xE8\x16`\xABq\xEB\xF2\xF3\xB4}\xB85\xAD\x08\xFF\x0C>!]\x9C\x01Iu\xD3A8\xFB\x8E\xB8\x16B\x0C\xB0!\xF7\x95\xBF/4\xAA\x89\xC8\xCF\x12\x12%G\xBC\xFB\xA9-\x1C\xE9\x1C^\xD5\xCA\x0E\xE7\xC7\xC6o\xD4\xD5\x89\xDA\x88\xC7p\x0EC\xCA\xE3\x18\x93-z\xA3\x01\x82\xCD\xA9\xF8\x0F\x86.1\xEA,\x05q\x0D\x0D-T\x91E\\\xD2r\xBD\x93\xE1!.>\x1C\xD1\xAFt^\xB1j\x99\xCF\xDB\x01\\\xF4\xD6h\x0E]\xA5\x8CR[\xFC5\xB1\xF9\xCA_\x14MmG\xA9W\xFD\x9F\xD2\x1A\x10c@\xD4w\x03\x81\x0Fr\xC3\x19\xF5\x15\xFB^\xDB\x15\xAE\xD3\xBE\xD8\xDB\xFB\xA8!=6\xB1\xE5\x92(\xAD\xA1\xFA\xB6;=\x03f\xD1!\x07S\x8A\xE1>2-\x0F-\xCC0C\x8B\xA6\x18J,$ \xB6\x0At\x13\x85\x15\xA6\xA48\x0A_IT\xD5\x09T\xA2`\xAD(\xAE"}\xFE\xCD\xD1\x9Ed\x08\xE4\x10\x00\xA4\xDF\x1D'\x08\xC2\x0F%>\x08\x9F\xFB\x96GZT\x89\x818&\xC6g5B\x86\x80\x0B\xB9\x98<\xD6\x9D\x9E\xC6(\x13oZ\xA5\xEC\xDC\x0E\xBEq\x02\x10$\xC5\xDB\xAAd\x80N\x9ES\xDDp\xFF\x8E\xE3#_\x06\x01\x02
>>> homekit_client_process: [Client 4] Finished processing
>>> homekit_client_process: [Client 4] Got 591 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 24748
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x03
[... 139 lines cut ...]
}
>>> homekit_server_on_update_characteristics: [Client 4] Processing element {
"aid":	1,
"iid":	12,
"ev":	true
}
>>> homekit_server_on_update_characteristics: [Client 4] Processing element {
"aid":	1,
"iid":	13,
"ev":	true
}
>>> homekit_server_on_update_characteristics: [Client 4] Processing element {
"aid":	1,
"iid":	14,
"ev":	true
}
>>> homekit_server_on_update_characteristics: [Client 4] Processing element {
"aid":	1,
"iid":	15,
"ev":	true
}
>>> homekit_server_on_update_characteristics: [Client 4] There were no processing errors, sending No Content response
>>> client_send: [Client 4] Sending payload: HTTP/1.1 204 No Content\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 4] Finished processing
```

peros550, 2020-08-06:
Sorry for asking, did you manually update esp-homekit after cloning and initializing repo and submodules, right?

beckmx, 2020-08-06:
@maximkulkin I have a similar issue, for some reason today, I tried to flash a brand new ESP8266(actually several I tried) and all of them seem to be failing to register, for some weird reason, if they connect to the "home" app they immediately change to disconnected, maaaaybe, the problem is that apple is no longer allowing new devices to log in, this is my log:

`mDNS announcement: Name=Temperature md=IR senderpv=1.0id=EE:08:AD:B6:BC:B2c#=1s#=1ff=0sf=0ci=5 Port=5556 TTL=4500
>>> HomeKit: [Client 3] Successfully paired
>>> HomeKit: [Client 3] Closing client connection
>>> HomeKit: Got new client connection: 3
>>> HomeKit: [Client 3] Pair Verify Step 1/2
>>> HomeKit: [Client 3] Pair Verify Step 2/2
>>> HomeKit: [Client 3] Found pairing with 78EBA429-C6E5-4E4D-B43E-606483BEEC27
>>> HomeKit: [Client 3] Verification successful, secure session established
>>> HomeKit: [Client 3] Get Accessories
>>> HomeKit: [Client 3] Remove Pairing
>>> HomeKit: Removed pairing with 78EBA429-C6E5-4E4D-B43E-606483BEEC27
>>> HomeKit: Last admin pairing was removed, enabling pair setup
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Temperature md=IR senderpv=1.0id=EE:08:AD:B6:BC:B2c#=1s#=1ff=0sf=1ci=5 Port=5556 TTL=4500
>>> HomeKit: [Client 3] Closing client connection `

Or is it because I haven't changed the pairing number?

maximkulkin, 2020-08-06:
@beckmx Can you enable HOMEKIT_DEBUG and post debug logs here. I would like to see what happened on that Get Accessories request.

beckmx, 2020-08-06:
@maximkulkin here u go

```
mode : sta(2c:f4:32:77:44:97)
Function called without core lock
Connected to WiFi
cws774497ADC start  
ADC read [3.000000] Temp [-71.066597] Rth [3400000.000000]
ADC correct [3.000000] Temp [-79.066597] Rth [3400000.000000]
ADC start  
ADC read [3.000000] Temp [-71.066597] Rth [3400000.000000]
ADC correct [3.000000] Temp [-79.066597] Rth [3400000.000000]
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x100000
>>> HomeKit: Generated new accessory ID: 1A:5D:4F:65:D8:03
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Temperature with IR-1A5D md=IR senderpv=1.0id=1A:5D:4F:65:D8:03c#=1s#=1ff=0sf=1ci=5 Port=5556 TTL=4500
>>> homekit_run_server: Staring HTTP server
>>> wifi_config: Stopping HTTP server
>>> wifi_config: Stopping DNS server

>>> HomeKit: Got new client connection: 3
>>> homekit_client_process: [Client 3] Got 148 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 31020
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 3] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 30924
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 3] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 30260
>>> homekit_server_on_pair_setup: [Client 3] Using user-specified password: 211-11-112
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> send_tlv_response: [Client 3] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): \xD2\xF5\xA4\xAE\x1D\xF7o\x81\xC7\x09\xFA.a\x10\xAD\xBE\x9F\x18r\x94R\xE3\xA9\x06(\x88\xD4.\xCC\x9D\x87l\xBC\x8CM\x1E\xB8*\x99~\x15)\x8A\xC7sY.\xD5@/\xCB\xA2\x09\x90\x0E[\xC5d\xACS\xD7\xBC\xCA\xE8\xC4VfD\x9B\xA0!\x91uS\xAC~\x06\xAD\xD3H\x87D\x9C>]L\x00Q\xF9,\x9C\xF5\xB1\x89\xEF\x936\xE4\xD1\xE6e\xB8\xDD={\x86\xE7\xD7\x1A\x19>?\xFA\xB4\\\x8C\x0D\xC4\x93\x94F{\xD8\xCE\xD3ql\xE2foe\xD8\xC35JJ\xB2E\x90[|\xFA\xA9\x9E\xCF|\xF2\x84\xB13G\x1B\xA5\xB5\x11\xE7\xDB\xA1\xDEg\xAA\xA5\x97\x80t\xC3k\xB7oV\x96\xE9;\x81\xDC\x0FB2\x9F\x8F\x88q\xD4\x0D\xD4^\xA0\x177nl|\xB7\x8DO\x9D\xF8\xDE\x05\x11\x16k5\x19\x84”\xEB\xCC\x80\xF9;pR\xB1\xDF\xF9V\xA1Bn!\xB5%D\x14\x08\xA3\x8A\x18\x07Do\xD2u\xF1\xDCw\x14\xFA~W\xDA8\x034\xBA\x0B\xD6/\x91q\x1E\x90\xA2\xE7\xC0\xA7\x97\xC3\xE1=\xAD\x1D\xB5Yo\xCDz\xBC\xB9\x8F\xB3\xD1\xCA\x83\xFB\x18.\x04\xDD\xC9%\xA3*\xDAB*Dow\xF3[\x19\x13\xC7\xE3\xAB>\xA4\xAE”`{\xD0=\x04*W\x00[\xA7\xCD\xE4>F\x9D\xBF\xEA\x03\xD3\x1C\x9E\x99\x17\x0E\xC9\xC2[\xAD\x86U\x9B\x13\x0C\xC6\xBF\x05B\x8F\x84\xB7\x8Fp\xFE\x11\x01K\x19QX\x87=\x13\xA0\x81\xA73\xB4VA6\xFF\x88\xB4\xA5\x1A\xF9,\x9AfM\x84D\x94f\xCD,\xEC\xF5*X@\xFB\xE4
>>> tlv_debug: Type 2 value (16 bytes): \xFD\x83(\x19\xA9\x98’\xE3F\x9BJ\x8B`i+7
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 3] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFF\xD2\xF5\xA4\xAE\x1D\xF7o\x81\xC7\x09\xFA.a\x10\xAD\xBE\x9F\x18r\x94R\xE3\xA9\x06(\x88\xD4.\xCC\x9D\x87l\xBC\x8CM\x1E\xB8*\x99~\x15)\x8A\xC7sY.\xD5@/\xCB\xA2\x09\x90\x0E[\xC5d\xACS\xD7\xBC\xCA\xE8\xC4VfD\x9B\xA0!\x91uS\xAC~\x06\xAD\xD3H\x87D\x9C>]L\x00Q\xF9,\x9C\xF5\xB1\x89\xEF\x936\xE4\xD1\xE6e\xB8\xDD={\x86\xE7\xD7\x1A\x19>?\xFA\xB4\\\x8C\x0D\xC4\x93\x94F{\xD8\xCE\xD3ql\xE2foe\xD8\xC35JJ\xB2E\x90[|\xFA\xA9\x9E\xCF|\xF2\x84\xB13G\x1B\xA5\xB5\x11\xE7\xDB\xA1\xDEg\xAA\xA5\x97\x80t\xC3k\xB7oV\x96\xE9;\x81\xDC\x0FB2\x9F\x8F\x88q\xD4\x0D\xD4^\xA0\x177nl|\xB7\x8DO\x9D\xF8\xDE\x05\x11\x16k5\x19\x84”\xEB\xCC\x80\xF9;pR\xB1\xDF\xF9V\xA1Bn!\xB5%D\x14\x08\xA3\x8A\x18\x07Do\xD2u\xF1\xDCw\x14\xFA~W\xDA8\x034\xBA\x0B\xD6/\x91q\x1E\x90\xA2\xE7\x03\x81\xC0\xA7\x97\xC3\xE1=\xAD\x1D\xB5Yo\xCDz\xBC\xB9\x8F\xB3\xD1\xCA\x83\xFB\x18.\x04\xDD\xC9%\xA3*\xDAB*Dow\xF3[\x19\x13\xC7\xE3\xAB>\xA4\xAE”`{\xD0=\x04*W\x00[\xA7\xCD\xE4>F\x9D\xBF\xEA\x03\xD3\x1C\x9E\x99\x17\x0E\xC9\xC2[\xAD\x86U\x9B\x13\x0C\xC6\xBF\x05B\x8F\x84\xB7\x8Fp\xFE\x11\x01K\x19QX\x87=\x13\xA0\x81\xA73\xB4VA6\xFF\x88\xB4\xA5\x1A\xF9,\x9AfM\x84D\x94f\xCD,\xEC\xF5*X@\xFB\xE4\x02\x10\xFD\x83(\x19\xA9\x98’\xE3F\x9BJ\x8B`i+7\x06\x01\x02
>>> homekit_client_process: [Client 3] Finished processing
>>> homekit_client_process: [Client 3] Got 601 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 28076
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x03
>>> tlv_debug: Type 3 value (384 bytes): \x18\xA9T?\xFFC\x85\x1C\x84\xC1\xA2\x14JU,\xA1\x89 \x15\x1FI\xFEJ\xEA\x03\xBA\xA6\xD1\xD7\xA1\xCAc\xF0|79\xD0{\xAFB\x92\xE2[-:\x16FaX\x039\xE8M\xF40\xC6\x8D\xFD\x0CB\x0B\xC1}\xDD\x038\x05\xC04\x04-A/_\x1Ba\xE0’\xEE\xBC\xB9\xB9\xA2_\xBC\xCDII\x877I\x0A)Wol\x8BW\xEF\xE9\xCD-\xE5\x99\x14SsakG\xDAH\xA0\x9Fl\x8C\xAC\xBAeo\x86\xBDEC\xC0\x0C\x0B\x01\x19t5\xDC/G\xF6\x0F\x98\xC5d\x17\xE5\x9BSo\x02\x0D3\xEB\x06\x18\x0A\x16\xAC\xE5\xE7\x87\x8D*\xE5An&(\xBF\xB7\xC3\xE6\xE0w\xD8\xDC\x12e\xB1M\xEE\x89%\xAA\x06v\xAA\x8E\xE7\xDC\xB1\x01)\xC4\x85\x16\x90b\x8C-B\xBC\xDD\xFC(U\xF3\xAA\x12\xD2\x8F{\x [... cut ...]
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 3] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 3\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x02
>>> homekit_client_process: [Client 3] Finished processing
>>> HomeKit: [Client 3] Closing client connection
ADC start  
ADC read [2.000000] Temp [-75.851006] Rth [5105000.000000]
ADC correct [2.000000] Temp [-83.851006] Rth [5105000.000000]

```

maccoylton, 2020-08-06:
> @maximkulkin I have a similar issue, for some reason today, I tried to flash a brand new ESP8266(actually several I tried) and all of them seem to be failing to register, for some weird reason, if [...]

@beckmx can you also try the Eve app and let us know if that works ok

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

RavenSystem, 2020-08-06:
Device pairs, but then it is removed, because temperature value (-78) is under min value (0), then it is an invalid value.

maximkulkin, 2020-08-06:
@beckmx ^^ what @RavenSystem said. Now let's figure out why your temperature value is not initialized. Can you send link to the code?

peros550, 2020-08-07:
There must be a generic issue. If you would like to reproduce the issue here what I did. I downloaded from scratch Maxim's esp-homekit-demo. I initialized the submodules and run the following command to bring sub-modules into their latest state: 

`git submodule foreach git pull origin master`

Then I setup wifi.h and tried the temperature sensor. Results are the same. The home app shows the accessory as "no response" when opening the room's page. Then if you click the accessory, it gets updated. 

Here is a [video](https://youtu.be/lbY60zyd7iA) showing the issue

Eve app shows the accessories as available and works properly. Issue found on ios 13.6 ,13.5.1 and iPadOS 14b4

peros550, 2020-08-08:
In order to isolate the issue, I made some tests in the temperature sensor example. I started applying all latest commits one by one. 
The issue appeared when I applied this [commit](https://github.com/maximkulkin/esp-homekit/commit/856f048cc3ddae8048f00a01e8e8209c577f9c46)

peros550, 2020-08-08:
Thanks Maxim! Today's updated fixed the issue! :)

closed 2020-08-08 by peros550