maximkulkin/esp-homekit #164: Request for advice: Optimizing code for better heap management

peros550, 2020-08-09:
This is not an issue with this remarkable library. Just a kind request for advice. 

Based on Maxim's examples, I am trying to consolidate various code examples into [one firmware ](https://github.com/peros550/esp-homekit-multiple-sensors/tree/master/examples/multiple_sensors_ac) to be used into Wemos D1 mini devices. The result would be a multi sensor device (motion, temp, hum, light). 

On top of that, I used Maxim's esp-ir to be able to send signals to the room's AC. All works great except from one thing, pairing. 

I understand that pairing process needs a lot of free heap to complete. 

In my troubleshooting, I tried in code to delay every init process I could, but it turns out my code use a lot of heap due to the need of RAW IR data, specifically in this [module](https://github.com/peros550/esp-homekit-multiple-sensors/blob/master/examples/multiple_sensors_ac/ac_commands.c)

Is there any way to overcome this issue? Would it be possible to not store this data in heap somehow?

maximkulkin, 2020-08-09:
I checked your firmware and indeed found out that raw IR commands consume RAM (~9-10K). First of all, I would recommend not to store commands in raw form but rather decode them into a nice 4-6 byte sequences. That would make much less headache.
But you decided to store the raw, so let's see what we can do there. It turns out that if you mark all your raw commands as "static const " (technically, you only need "const", but "static" also makes sense to do), then compiler will not put those into RAM and you will get your memory usage down to a manageable values (30K RAM free after HomeKit initialization). That should solve your pairing issues.

Hope that helps.

peros550, 2020-08-10:
Many thanks Maxim! Your suggestion worked very well.

I had considered decoding them, but it wasn't very easy for me. I think esp-ir does provide a decoding function, I will try to check that again.
 
One more question in regards to building the project. I have noticed that making a few changes in the main.c file and trying build command make with "test" attribute does not rebuild the whole project. I guess this is a safe approach during development. 

In this way, you keep storage unaffected right?  

When I want to update a device and include updates from sub-modules, is it still safe to try make command with test attribute?

In general, in which case, is best to use erase_flash command over test ?

pbendersky, 2020-08-10:
@peros550 [this Arduino library](https://github.com/ToniA/arduino-heatpumpir) has many supported ACs (probably yours is supported there) and has easy to read code that you can probably port, instead of having your raw commands. If you encode your command, you'll save a lot of heap, and end with a cleaner code.

peros550, 2020-08-27:
Many thanks for all advices !!!

closed 2020-08-27 by peros550