maximkulkin/esp-homekit #167: mDNS: Got potential name conflict, waiting 1s to retry and confirm

peros550, 2020-08-10:
Updated to the latest version, and after a few reboots I started to receive the following messages: 

(two different devices , running same firmware)

```
mDNS: Got potential name conflict, waiting 1s to retry and confirm
mDNS: Probing 1 HomekitSensor-59EF962DCF46
mDNS: Got potential name conflict, waiting 1s to retry and confirm
mDNS: Probing 1 HomekitSensor-59EF962DCF46
mDNS: Got potential name conflict, waiting 1s to retry and confirm
mDNS: Probing 1 HomekitSensor-59EF962DCF46
mDNS: Got potential name conflict, waiting 1s to retry and confirm
mDNS: Probing 1 HomekitSensor-59EF962DCF46
mDNS: Got potential name conflict, waiting 1s to retry and confirm
mDNS: Probing 1 HomekitSensor-59EF962DCF46
mDNS: Got potential name conflict, waiting 1s to retry and confirm
```
and 

```
mDNS: Probing 1 HomekitSensor-ABAB862DCF46
mDNS: Got potential name conflict, waiting 1s to retry and confirm
mDNS: Probing 1 HomekitSensor-ABAB862DCF46
mDNS: Got potential name conflict, waiting 1s to retry and confirm
mDNS: Probing 1 HomekitSensor-ABAB862DCF46
mDNS: Got potential name conflict, waiting 1s to retry and confirm
```
Am I doing something wrong? Any idea on how to troubleshoot?

peros550, 2020-08-19:
@maximkulkin may I ask for your help?
The above issue must have been resolved in one of the recent commits. 

I'm having a different issue right after I reboot a device which stays in no response status. This issue does not appear in every reboot. It appears randomly. 

Here is a log when the device did not boot properly: 

```
Opening /dev/ttyUSB0 at 115200bps...
ip:192.168.0.142,mask:255.255.255.0,gw:192.168.0.1
>>> wifi_config: Connected to WiFi network
CONNECTED TO >>> WIFI <<<
HOSTNAME set>>>>>:HomekitSensor-59EF96
manuf='manuf_unknown' serial='5C:CF:7F:59:EF:96' model='model_unknown' revision='0.0.0' c#=0
Free Heap: 34392
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 09:3E:78:E4:1A:AD
>>> HomeKit: Found admin pairing with 0C7EC359-6CF6-489F-A43A-597E97DAA140, disabling pair setup
>>> HomeKit: Configuring mDNS
HomeKit: adding mDNS TXT record md=model_unknown
HomeKit: adding mDNS TXT record pv=1.0
HomeKit: adding mDNS TXT record id=09:3E:78:E4:1A:AD
HomeKit: adding mDNS TXT record c#=1
HomeKit: adding mDNS TXT record s#=1
HomeKit: adding mDNS TXT record ff=0
HomeKit: adding mDNS TXT record sf=0
HomeKit: adding mDNS TXT record ci=9
SERVER JUST INITIALIZED
Started UDP logging1
Found pairing, starting timers
Initializing AC parameters
Movement not detected
>>> homekit_characteristic_notify: Got characteristic 2.19 change event
Temp:23
mDNS: Delaying probe until IP is acquired
>>> homekit_run_server: Starting HTTP server
Got IP, resuming mDNS
mDNS: Probing 1 HomekitSensor-59EF96
mDNS: Failed to send broadcast (code 118)
mDNS: Probing 1tSensor-59EF96
mDNS: Failed to ip:192.168.0.142,mask:255.255.255.0,gw:192.168.0.1
mDNS: Probing 2 HomekitSensor-59EF96
mDNS: Probing 3 HomekitSensor-59EF96
mDNS: Probe successful, announcing HomekitSensor-59EF96 TXT md=model_unknown pv=1.0 id=09:3E:78:E4:1A:AD c#=1 s#=1 ff=0 sf=0 ci=9
mDNS: Announcing 1 HomekitSensor-59EF96
mDNS: Announcing 2 HomekitSensor-59EF96
>>> homekit_characteristic_notify: Got characteristic 1.11 change event
>>> homekit_characteristic_notify: Got characteristic 2.13 change event
>>> homekit_characteristic_notify: Got characteristic 2.22 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
mDNS: Announcing 3 HomekitSensor-59EF96
>>> homekit_characteristic_notify: Got characteristic 2.22 change event
mDNS: Announcing 4 HomekitSensor-59EF96
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.22 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.22 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 1.11 change event
>>> homekit_characteristic_notify: Got characteristic 2.13 change event
>>> homekit_characteristic_notify: Got characteristic 2.22 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 1.11 change event
>>> homekit_characteristic_notify: Got characteristic 2.13 change event
>>> homekit_characteristic_notify: Got characteristic 2.22 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
RC !!! ERROR Sensor
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 1.11 change event
>>> homekit_characteristic_notify: Got characteristic 2.13 change event
>>> homekit_characteristic_notify: Got characteristic 2.22 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 1.11 change event
>>> homekit_characteristic_notify: Got characteristic 2.13 change event
>>> homekit_characteristic_notify: Got characteristic 2.22 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.22 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.22 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.16 change event
>>> homekit_characteristic_notify: Got characteristic 2.22 change event
```

Any ideas for me to troubleshoot further?

maximkulkin, 2020-08-20:
I do not see (at least from logs provided) that this device did not boot properly. I see that device has started and announcing itself over mDNS. On top of that it looks like there is a periodic poll for sensor data updates which then do homekit_characteristic_notify().

closed 2020-09-17 by peros550