maximkulkin/esp-homekit #169: Предложение идеи проекта

exrector, 2020-08-20:
Добрый день, Максим!
Так как вы у нас являетесь основоположником, хотел вам предложить идею, для которой у меня не хватает знаний и опыта. 
Исходные данные:
Используются платы esp, sdk espressif и emulator HomeKit devices
Идея:
Так как мы используем всегда одинаковые платы и можем унифицировать их, то мы могли бы создавать универсальное устройство согласно встроенным сервисам apple. 
Создаётся универсальная прошивка с жесткой привязкой ко всем gpios
В emulator HomeKit devices (стандартное средство apple доступное любому) создаётся проект устройства и сохраняется в файл. 
Далее этот файл (или его содержимое в виде кода)нужно как-то внедрить в универсальную прошивку. 
Тогда остаётся только покупать платы esp. Прошивать. И подключать периферию (моторы, датчики, реле и пр) к заведомо выделенным gpios. Все остальное типа оформления, qr, кода устройства и пр берет на себя эмулятор.
Что думаете?

maximkulkin, 2020-08-20:
Доброго времени суток. Идея универсальной прошивки не нова, но меня всегда больше интересовало фичи и оптимизация фреймворка, а не поддержка миллиона одинаковых устройств. На базе моей библиотеки есть сторонние решения, которые решают те проблемы (например [HAA](https://github.com/RavenSystem/haa)). Хотя последнее время я тоже приступил к разработке своей универсальной прошивки, но она еще не готова.

exrector, 2020-08-20:
> Доброго времени суток. Идея универсальной прошивки не нова, но меня всегда больше интересовало фичи и оптимизация фреймворка, а не поддержка миллиона одинаковых устройств. На базе моей библиотеки [...]

Да, я знаю про RavevCore haa. Конечно я их использую, потому что нет альтернативы. 
В своей идее я несу посыл ни то что б за универсальность прошивок, сколько за использование HomeKit accessory emulator как базовый инструмент, в котором можно реализовать любое устройство. Это очень захватывает. А подстройка одним только json несколько ограничена. Увы. 
Я с вами согласен про миллион одинаковых устройств. Об этом и речь. Я игрался с эмулятором и я вижу, сколько интересных устройств можно сделать. А так как официальный sdk HomeKit есть только для esp32  то поэтому я заинтересовался открывшейся возможностью создать для себя интересный проект. 
Тем не менее, спасибо за ответ и инсайдерские новости. С нетерпением буду ждать ваших разработок.

maximkulkin, 2020-08-20:
А что особенного в использовании симулятора?! Он позволяет только накликать сервисов и характеристик в свое устройство. То же самое можно сделать просто открыв спецификацию HAP (или почитать комменты в include/homekit/characteristics.h. Это не сказать что главная проблема создания прошивки.

maximkulkin, 2020-08-20:
> Да, я знаю про RavevCore haa. Конечно я их использую, потому что нет альтернативы.

1. RavenCore - это другой проект того же автора, который подписывается как RavenSystem.
2. Альтернатива - писать кастомные прошивки под конкретные устройства (как это делаю я). В универсальных прошивках хочется впихнуть много, а ресурсов - мало. Аппетиты растут очень быстро, а потом люди шлют репорты, что "как же так?! память закончилась, почините".

exrector, 2020-08-20:
> 2\. Альтернатива
Ни в коем случае не подумайте, что я спорю с вами только, пожалуйста. Вы мой бесспорный кумир в этом вопросе. И мне важно ваше мнение. Поэтому я и хотел бы у вас научится и предлагаю своё видение на ваш суд. 
Мой мозг просто устроен в векторе унификации и оптимизации. Отсюда идеи)) Так что вы уж простите за некоторые пробелы в знаниях. 

Вы пишете про конкретные устройства. А что это? Можем ли мы определиться?
1.Это устройства и их характеристики, которые заложены apple в hap?
2. Устройства представленные производителями?
3. Чипы для diy?

Что мы хотим? Мы хотим быть независимы от производителя и его набора характеристик? Например есть девайсы в которых есть возможности, но они скрыты, потому что можно продать дополнительный гаджет. Профит. 

Поэтому для себя я пришёл к выводу, что используя универсальный esp и его открытый sdk homekit можно сделать болванку https://github.com/espressif/esp-apple-homekit-adk. Сервисы и характеристики устройств в спецификации ограничены и их все можно унифицировать. 
Просто пример, чтобы вы оценили элегантность подхода:
Я бeру esp32. Устанавливаю болванку прошивки, которая под разные характеристики выделяет в коде конкретные gpio. Накликиваем в эмуляторе, например лампочку и сохранённый файл экспортируем через веб интерфейс в прошивку. Готово устройство. Если нужно задействовать дополнительные gpio мы берём сохранённый файл, накликиваем ещё и датчик движения. Делаем импорт через веб интерфейс. Бум. Готово. Осталось подключить сам датчик. Прелесть в том, что человек в графической форме визуально может создать устройство. Причём предварительно проэмулировав его перед прошивкой на своём телефоне в приложении home. Все id, специфические названия и ссылки взаимозависимостей. Все это делает эмулятор и сохраняет в файл. Вот что ценно, мне кажется. Я сделал этот вывод основываясь только на том, что xcode использует этот файл для проекта. 
Но я могу ошибаться и, наверное, делюсь с вами, чтобы вы либо заразились такой унификацией, либо окончательно разбили в пух и прах мой подход, чтобы я начал думать в другом направлении.
И спасибо, что делитесь и раскладываете по полочкам для меня. Мне очень интересно. Правда.

kistanod, 2020-09-05:
Я увидел как через облако можно менять характеристки устройств esp с помощью esp rainmaker. Вот ссылка https://rainmaker.espressif.com/ 

Это идея пересекается с тем, что описывает @exrector.

Возможно ли объеденить эти две идеи?

maximkulkin, 2020-09-05:
@exrector я читаю и у меня создается впечатление, что Вы не пробовали перепрошивать конкретные устройства. Никаких "скрытых функций" чаще всего нигде нет, это просто не выгодно впаивать компоненты, которые не будут использоваться. Производители максимум что делают - это делают печатные платы под несколько похожих устройств, но компоненты впаивают только те, которые нужны конкретному устройству. Поэтому обновление прошивки не превратит лампочку в термостат.

Что чаще всего происходит - это Вы покупаете какую-то китайщину со своей прошивкой, работающей с облаком конкретно этого производителя (и сомнительными практиками информационной безопасности), управляемое кривым приложением для телефона. Вы перепрошиваете это своей прошивкой, которое ни в какие облака не ходит и работает через инфраструктуру HomeKit. У каждого устройства есть своя специфика (кроме совсем простых типа "включил/выключил"). Например, у меня есть устройство, которое управляет кондиционером по ИК.. ему надо понимать команды ИК и уметь транслировать их в состояние HomeKit термостата. Чтобы такое сделать из универсальной прошивки, Вам надо будет нечто большее, чем просто загрузить описание устройства из симулятора. Надо иметь некоторое подобие скриптов, которые надо будет интерпретировать прямо на устройстве.

Также, ваше описание устройства из симулятора не будет иметь информации о том, что делать, если значение характеристики ON вдруг изменится, какой gpio дергать и т.п. Также, оно не будет знать каким протоколом общаться с очередным сенсором, впаяном в устройстве. Это все нужно программировать.

Для простых устройств, где не нужно кудрявой логики, уже есть HAA, который я упоминал выше. Делать супер-пупер прошивку со всеми драйверами под все сенсоры и устройства - идея, не лишенная смысла, но очень трудозатратная. Но переиспользовать описания из симулятора - это решение лишь маленькой проблемы, которое для многих станет еще и дополнительным барьером (ибо надо будет качать еще одно приложение, да еще только под Мак).

closed 2020-09-14 by maximkulkin