maximkulkin/esp-homekit #175: mDNS: Failed to send broadcast (code 118)

peros550, 2020-10-07:
Any idea why this is happening?

```
mDNS: Probing 1 HomekitSensor-7E662C
mDNS: Failed to send broadcast (code 118)
mDNS: Probing 1 HomekitSensor-7E662C
mDNS: Failed to send broadcast (code 118)
mDNS: Probing 2 HomekitSensor-7E662C
mDNS: Failed to send broadcast (code 118)
mDNS: Probing 3 HomekitSensor-7E662C
mDNS: Failed to send broadcast (code 118)
mDNS: Probe successful, announcing HomekitSensor-7E662C TXT md=esp-homekit-ac-daikin pv=1.0 id=A8:11:C5:28:0F:D3 c#=2 s#=1 ff=0 sf=1 ci=9
mDNS: Failed to send broadcast (code 118)
mDNS: Announcing 1 HomekitSensor-7E662C
mDNS: Failed to send broadcast (code 118)
mDNS: Announcing 2 HomekitSensor-7E662C
mDNS: Failed to send broadcast (code 118)
mDNS: Announcing 3 HomekitSensor-7E662C
mDNS: Failed to send broadcast (code 118)
mDNS: Announcing 4 HomekitSensor-7E662C
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
mDNS: Failed to send broadcast (code 118)
```

maximkulkin, 2020-10-07:
Haven't seen that problem. Are you sure it is connected to a network? Maybe network got down/haven't connected yet, but for some reason mDNS was still started?

peros550, 2020-10-07:
Thanks for your quick feedback!
This happened multiple times right after Server Initialization event, right after a restart. I tried with another wemos d1 mini and same results. I wasn't able to add the device in homekit, it wasn't showing up and if it was, after a lot of waiting, it couldn't complete pairing, stuck on step 1. 

Same device on my home network works perfectly fine. Therefore I can only blame the router (TP Link Archer VR400). 

For the record, another HAA device which coexists on same LAN seems to work perfectly fine. 

I will continue troubleshooting and let you know.

peros550, 2020-11-02:
@maximkulkin once again I can't thank you enough. 

Today I had the chance to visit my sister and troubleshoot the issue further. Your tip gave me the direction where to look for issues. Although the device seemed to be responsive in pings some of the time (not 100% of the time) it definitely felt something was wrong with the network. I tried to remove auto channel on the router (TP-LINK VR400) and set a random one and the issue immediately cleared. 

I wish you all the best
Many thanks for your help!

closed 2020-11-02 by peros550

maximkulkin, 2020-11-02:
I started to think that some unresponsiveness of ESP8266 accessories come from WiFi stack, not from HomeKit implementation. Sadly those are much harder to troubleshoot, because WiFi stack implementation is closed source.