maximkulkin/esp-homekit #178: Timer Switch option?

Maarten-s92, 2020-11-09:
Hi,

Maybe I’m not looking in the right place, but I can’t find any option to set a or program a timer. 

I use the esp as a switch for my audio, but my girlfriend forgets to turn it off. So having a timer would be a great solution. Anybody that can help me?

Thanks a lot in advance!

maximkulkin, 2020-11-10:
Could you be more specific on what you're trying to achieve? For example, if you want to create an automation that will turn off a switch after 1 hour, you can do it with automations and new Shortcuts app:
* Go to automations and press "+"
* Select "Create Home Automation"
* Choose "An Accessory is Controlled"
* Pick you switch and press "Next"
* Select "Turns On", press "Next"
* On next screen, scroll all the way down and in "Advanced" section press "Convert To Shortcut"
* In Shortcut screen, press "+" button, select "Scripting" and select "Number".
* In Number block enter 3600 (number of seconds to wait before turning switch off).
* Press "+" button again and add "Wait" block.
* Long press on placeholder in "Wait" block, choose "Select Magic Variable" from popup menu and select your Number block.
* Drag initial HomeKit "Set" block after "Wait" block, click on it's placeholder "Scenes and Accessories", select your switch, press Next and select "Off" state.
* Finish creating automation.

Now it will automatically turn off after an hour after it was turned on.

Maarten-s92, 2020-11-11:
Thanks A lot for the tip and the great discription Maxim. 

I tried to use your set-up and got it working for a minuten than I made some changer and it all went downhill from there. 
The thing I would like is that the moment someone turns on the switch trough their iPhone the timer starts running. The problem I now have is that I can activate the timer but only trough the automation it’s self. I got some succes with scènes but somehow I could not get it working fully automatic. Maybe it’s the problem that I don’t have a HomeKit hub. I tried using my iPad mini2 as one but it only has iOS 12 which doesn’t allow me to use the option “an accessory is controlled”. I have an old iPhone which I maybe could jailbrake to iOS13 so that will be the next thing to try. 

Thanks a lot!

maximkulkin, 2020-11-11:
Oh well, you do need a hub to be able to do HomeKit automations.

Maarten-s92, 2020-11-11:
Ah that’s the thing. Isn’t it possible to “build” a hub? I’m not sure iff I want te voice control stuff

maximkulkin, 2020-11-11:
No, you can not build a hub.

Maarten-s92, 2020-11-15:
Hi Maxim,

So i tried to make 2 hub's one with my ipad (Ios12) and one with and old iPhone jailbroken on ios 13. In bots cases I can't select "an accessory is controlled" it's grayed out. Do you maybe know why that would be?

Maarten-s92, 2020-11-17:
After adding a homebridge for some other sockets it works now! So thanks for your tips!

yeffal, 2021-02-21:
Hi Maxim,

Is there a way that I can ask Siri (In homekit) turn the switch (any switch) for 10 minutes?
I got what you explain in:"Could you be more specific on what you're trying to achieve? For example, if you want to create an automation that will turn off a switch after 1 hour, you can do it with automations and new Shortcuts app:"

But I do not want to turn off always in 10 minutes, maybe this time a just want to turn off, but in other time to turn on and turn off after 10 minutes or any timer at that time. 

I can do this very easy with Alexa and I´m trying to just use siri. 

Thanks a lot for your help.