maximkulkin/esp-homekit #180: Question: When to use a separate task

jpasqua, 2020-11-17:
I haven't been able to get comfortable with when it is appropriate / necessary to perform an action on a separate task vs. inline. For example, in the button example, imagine you wanted to blink an LED for user feedback when a button is pressed. Is it OK to perform the IO and delays in the button callback, or is a separate thread necessary?

Thanks for any input/guidance.

maximkulkin, 2021-01-04:
I would not recommend doing anything like that in button callbacks, although sometimes a little bit of console output can be tolerated. Otherwise you need some kind of queue or a eventgroup to signal events from button callbacks. Also, depending on button library: some libraries call callbacks right from interrupt handler, some use timers (and thus run inside timer task). Spending time in interrupt handler might cause GPIO changes missed and thus bad presses/releases registration.

closed 2021-01-04 by maximkulkin

jpasqua, 2021-01-05:
After digging more deeply I cam to that conclusion. Thank you for confirming.