maximkulkin/esp-homekit #182: Connecting with QR code

emresenyuva, 2020-11-27:
why wifi info is needing?, can we connect with just QR code?

renandw, 2020-11-27:
This firmware is using HAP (HomeKit accessory protocol) over IP so Iit needs constant Wi-Fi connection.

maximkulkin, 2020-11-27:
MFi certified accessories (commercial ones) can use Apple's special onboarding procedure which can find accessories and configure them to join your WiFi network. Unfortunately, non-commercial accessories can not do that.

konifer44, 2020-12-11:
Im using
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/wifi_provisioning.html
 to send WiFI Credencial Via ESP SoftAP Prov
So you no longer need to store WiFi password hardcoded 😊

maximkulkin, 2020-12-18:
@konifer44 That's a nice discovery for those of us who run ESP32 (not sure if this is available for ESP8266_RTOS_SDK). For onboarding I'm actually using an [esp-wifi-config](https://github.com/maximkulkin/esp-wifi-config/) library that I built. Although it is still not as slick as just scanning a QR code, it allows not having hardcoded WiFi credentials inside firmware.

konifer44, 2020-12-19:
It's very fast and convenient but I had to add button to reset WiFi credentials if you want connect to another network 😊
Here is my switch with my own PCB
![Uploading 102681540...]

closed 2021-01-04 by maximkulkin