maximkulkin/esp-homekit #184: Can't pair on current master (stack on 1/3 and closes connection)

nonameplum, 2021-01-03:
Hey @maximkulkin,
I have updated my repo to use current master
https://github.com/maximkulkin/esp-homekit/commit/379f593baad364c90233587a96241f636719d103
and I can't get a device paired in comparison to 
https://github.com/maximkulkin/esp-homekit/commit/0f8dd68bc1499f38f82771ac58ddaf3c778cd87f
that I lastly used and worked fine

I did the same process using both builds (erased device with first pairing) and compared the logs (I will paste probably the crucial difference):

The repeating behaviour that I found (on master) is that :
```
>>> homekit_client_process: [Client 1] Finished processing
>>> HomeKit: [Client 1] Closing client connection from 192.168.0.107
```

I have checked and compared `server.c` changes but here is a lot of changes and I'm not sure if this is related to the changes made in the method `homekit_client_process`.

If you would like me to put more logs or details let me know.
I would love to get it running as I'm quite exited to try out the new MDNS implementation.
Btw I also see more heap used in the new implementation.

## Logs

### Old one (working)
```
dhcp client start...
ip:192.168.0.119,mask:255.255.255.0,gw:192.168.0.1
+++ 23:59:59 [on_wifi_ready] [wifi_setup.c:56] Starting TFTP server...
+++ 23:59:59 [wifi_connected_handler] [garage.c:364] Homekit server init
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x100000
>>> HomeKit: Generated new accessory ID: C6:65:0D:B8:EB:CA
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=GarageDoorTest  md=PLDO02pv=1.0id=C6:65:0D:B8:EB:CAc#=1s#=1ff=0sf=1ci=4 Port=5556 TTL=4500
>>> homekit_run_server: Staring HTTP server
>>> HomeKit: Got new client connection: 4 from 192.168.0.107
>>> homekit_client_process: [Client 4] Got 132 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 34268
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 34176
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 4] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 33512
>>> homekit_server_on_pair_setup: [Client 4] Using user-specified password: 111-11-987
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> send_tlv_response: [Client 4] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): \x1C\xAD\xBC\xA9\xC6e\xBB\xC12>\xF2M,-"\xC0\x92Fjz\x1B\x1A\x86\x0D\xBA\xC798\x92\x17y\x1D\xD1F\xD2\x95h\xEDD\x08\xA3\xC2\x15\xCB=1\x93\xF9n\x06/+\xB6t\x11\x03f\x9E\x8E\x81\x9D\xDF\x06\x8F\xF1=e:\x8EW\xDA\x01\x0FI\xD0q\x883\xED\x05\xCDZp\xB5\xB6\x98\xE4\xD6&Wy\x14\xF8\xFD&\xE2\xDEG?\xB52eJ~\xE6\xF2}\x09Or\x98\x81\x8A\xAB<w\xDAf\xDE#\x01\x8E)\x0An\x04\x80\x09\\n\x06\x83\x90|2J\xA2\xFAb\xC4OXqX\xAB\xB90 \xCC$\xE6[\x90(\x90\x19\x8C\xFB\xF0\xF7\xCA2Y%\xF3N\xAD9\x18a_\xBD\xC7\xB2\x0C\x00]/d\xC9b-S\x83\xA6[O\x8D}\xBC\x81\xF6\xA6"w\xAB\x98\xA0\xEE\x03\xF1\xB0\xB6\xF0\x1By\x18\xE9\xB8\xE4\x0F\x17d\xD6 \xD0\xCD\xCB\xE2\xB8\xD0\x0A\x157)\xF3\x06 \xF9\xD9\x9E\x84;-k\x1CB\x8B\xB4\x8D\x84]\xF9\x8E\x9D\xCFl\x19\xD7\xF4\x1A <\xDC\xD8cE?a\x12\xA5_\x16\x06\xBA\x0A\x00\x17\x0A\x0F\xCE\xF8\xF2\x19u\xBD-{l\x81H\xE9\x9D\x8B+x0\x16q\x85\x9D\x0B\x94n~j\x00&\\p\xD6\x93\x01.4{\x0F\xA1\xAB\x9B\x1B\x95\x88\x9B\xEET\xD8>\xE5<\x8C\xFF\x94\x1B\x07\xCD+\xC6>\xFD\xAE!\x85\xBA\xE7\x94lu\x0C\xA4\x9Cj\xD9\x9A7J&\x01\x93\xCC\xCD\x9A\xE4t5K\xCB+\x0A\xB8%K>\xC3\xA9Jx\x99\xCB\xDC<\xDC%\xB4S\xBC\x18\x81\xC9\xC2\xDB\xAFun
>>> tlv_debug: Type 2 value (16 bytes): \xB2\x04\xDF\xA1\xBD8$\xF6,bi\x99oE\x02\x0B
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 4] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFF\x1C\xAD\xBC\xA9\xC6e\xBB\xC12>\xF2M,-"\xC0\x92Fjz\x1B\x1A\x86\x0D\xBA\xC798\x92\x17y\x1D\xD1F\xD2\x95h\xEDD\x08\xA3\xC2\x15\xCB=1\x93\xF9n\x06/+\xB6t\x11\x03f\x9E\x8E\x81\x9D\xDF\x06\x8F\xF1=e:\x8EW\xDA\x01\x0FI\xD0q\x883\xED\x05\xCDZp\xB5\xB6\x98\xE4\xD6&Wy\x14\xF8\xFD&\xE2\xDEG?\xB52eJ~\xE6\xF2}\x09Or\x98\x81\x8A\xAB<w\xDAf\xDE#\x01\x8E)\x0An\x04\x80\x09\\n\x06\x83\x90|2J\xA2\xFAb\xC4OXqX\xAB\xB90 \xCC$\xE6[\x90(\x90\x19\x8C\xFB\xF0\xF7\xCA2Y%\xF3N\xAD9\x18a_\xBD\xC7\xB2\x0C\x00]/d\xC9b-S\x83\xA6[O\x8D}\xBC\x81\xF6\xA6"w\xAB\x98\xA0\xEE\x03\xF1\xB0\xB6\xF0\x1By\x18\xE9\xB8\xE4\x0F\x17d\xD6 \xD0\xCD\xCB\xE2\xB8\xD0\x0A\x157)\xF3\x06 \xF9\xD9\x9E\x84;-k\x1CB\x8B\xB4\x8D\x84]\xF9\x8E\x9D\xCFl\x19\xD7\xF4\x1A <\xDC\xD8\x03\x81cE?a\x12\xA5_\x16\x06\xBA\x0A\x00\x17\x0A\x0F\xCE\xF8\xF2\x19u\xBD-{l\x81H\xE9\x9D\x8B+x0\x16q\x85\x9D\x0B\x94n~j\x00&\\p\xD6\x93\x01.4{\x0F\xA1\xAB\x9B\x1B\x95\x88\x9B\xEET\xD8>\xE5<\x8C\xFF\x94\x1B\x07\xCD+\xC6>\xFD\xAE!\x85\xBA\xE7\x94lu\x0C\xA4\x9Cj\xD9\x9A7J&\x01\x93\xCC\xCD\x9A\xE4t5K\xCB+\x0A\xB8%K>\xC3\xA9Jx\x99\xCB\xDC<\xDC%\xB4S\xBC\x18\x81\xC9\xC2\xDB\xAFun\x02\x10\xB2\x04\xDF\xA1\xBD8$\xF6,bi\x99oE\x02\x0B\x06\x01\x02
>>> homekit_client_process: [Client 4] Finished processing
>>> homekit_client_process: [Client 4] Got 585 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 31324
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x03
>>> tlv_debug: Type 3 value (384 bytes): \xF2\xC8\xB5\x82\xB8v\x10Y\xE9\x88\xB0\x81'\xFDJd\xAB\xC9Sl;\xBCe\xA9\x14f3~ \x88\x84*\xEA\xCC\x9B \xA2\xFE5\xDC\xD2\xB9~&\xB1C\xFE$j\x94u\xEC\x08 8\xC5\xC0\x93\x1B\xA4!\xD7\xC2\x80\xC1cm\xF0?\x0A\xD7H\x154\xE3\x16\x05<\x18\x9Fm\xF3\xBB|C3M\xBB\x02\xB3\xE0\xFC\xDE\x0D/B\x96]\xF7v\xCF\xAB\xDA\xEB\xF5kl\x92pCo/\x04\x18z\xDD4\xC1\xB6J[\xE5\x80y\x93O\xBD&\x1BQ\x10\xE8\xA9E\xC7*\xFCu)\x02<Q\xA0J\x98\xBE\x98\xFBs\x0B\x82\x04_\xC35}\x89\xAE\x9B\xA3\xBB\xC4-\x99>\xCAEV\xF6N\x8F!\xC5(\xFB\x17#H\xE7]\x1Ay\xDB\x9D\xB8\x9DI\xB4\x18\xC1F\xB4d%\xD9\xB4oL\x0E\xE8\x80D\x0B\x90\xCC\xC7R\xE06\x83<\x17\x8A\xAD\x12\\\x05\xE5iFo-\xCFI\xE7f\xE2\x0B\xB4Q\x0A\xD6.\x1B\x99\xEC\x19\xB6\x1D%v\xC9+%\x1E_\x95\x0D\x8A\xA8\xB6QW?\x1F\xF3\xE3\xBA\\?)R[b\x8F7\xAD"'\x1B0E/\x88N\xEB|S\xA1\x08\x03T7\xD5FY\xC7\xEC\x0D/\x97\x9Fx\xC1\xD6\x9C\xB0G\xBE\\I'D$3\xC6\x99\xF2\x1CA\x8E\x86\xD3\x86\xF8\xFE\xE11;4\xDC\xC4W6\x93\xB5\x0Dc}\xCE\xA4y]\xE0\xEC\xB03k\xF6\x9DF\xBB\xB3f\x13\x87\xC5\xE9L\xE4=\x03*\xD2\x94fmu\xDE\x13\xEF\x12%\xCE@l\x91\xDF%q\xF1XR\xBB\xC0\xC0\x93\xC9\x89his
>>> tlv_debug: Type 4 value (64 bytes): \x0E\xD0\xE4\x93\x08h\x9E\x1B\x16\x07p\xEA"C{\xB8\x96Q\x17\x89\xF2\x8C6\xB9\x8E\xF0\x13\xF3\xDCc\xF1j\xEC\xC3\x082\xA2\xCC\xDB6\xE3\x0B\x9E\x0DZX\xF5\xC8\xB8)w\xC0Z!U\x9F\xBD!\xC6\x04%~n\x13
[... 73 lines cut ...]
>>> homekit_server_on_pair_setup: Free heap: 28548
>>> homekit_server_on_pair_setup: [Client 1] Using user-specified password: 111-11-987
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> client_sendv: [Client 1] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0ATransfer-Encoding: chunked\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: 199\x0D\x0A\x03\xFF\xA2\xD5 \xB8Ej\xD9\x18\xBAp\x0D\xAFK\xB7\xB8\xEB\xAE\xCF;^=\xF4\xA4l\xE2\xA1\x8E\x17L^8S\x8D\xA9\xB6\xAB\xEE.;\xAF\xC63u\xF3\x16\xCD\xEA\xC8\xB3m\xA1\x8F\xE6Nt\xF1 \x0D\xE2G~gj\xD1M\x80w1U\x0D\xED\xFE\xBC\xE6\xDE\x96\x0A\xB6\xAD\x04\x9E\xE0Q\xF8\xAE\xDCu\xD2\x15\xDE\x9Aw\x8B\xD9\xCF\xC3*\x0A\xD1\xAE\xCC\x0B\x1E\x1Cz0\x98z\x0E<@\x95\xF23{\x98\x18\x18\xF1\xCA\x08M#\x81\xA5\xED\x93q\xE9\xE43\x01\x8FR<\xC0\xAAU\xA8RM\xC0!\xA3_\x8A\xCE\x8BG\x87I\xC6\xCB\x9D7\xE7\x1A\xC9\x92\xCEK\x12\x0F\xFA\x84\xC8\xDAw\xD4\x81\xE7~1\x93\xF3\\\xD82"G\xB2\x15h\xD2W\xDDk\x13\x86\x96\x05u\xF1\x95Eh\xF9\xB7\x90\x8FV\x07+\x82rM\x80\x19\xDA=\x03\x06l0\xD9\xAC\xDF\xD0u\xFE$\x1F\xC9\x9A\xCDR\xAB[\xBEPo\xBA3\xA6pb\xF5\xAA\xFAb\x8C\xF7\xF8\xA7\x14\xDC,G\x91\xEA\xA5~\x0Da,\x03\x81\x90f\x8D\x9B\xAA"\xF2H\xCEN\x896\x [... cut ...]
>>> HomeKit: [Client 1] Got new client connection from 192.168.0.107
>>> homekit_client_process: [Client 1] Got 132 incoming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 29312
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1] Pair Setup Step 1/3
... (it will repeat few times the same unit the Home app on the iOS device will say that it failed to pair)
```

nonameplum, 2021-01-04:
@maximkulkin It seems like the issue is present only on the last commit
https://github.com/maximkulkin/esp-homekit/commit/379f593baad364c90233587a96241f636719d103

The issue that I observed is that once the server does the first `homekit_client_process` and makes the `Calculating public key` and sends the payload it gets stuck because then in `homekit_run_server` -> `select(server->max_fd + 1, &read_fds, NULL, NULL, &timeout);` returns all the time `0`. It keeps do "nothing" until the client is disconnected and start over again.

maximkulkin, 2021-01-04:
That's not what's happening. Last commit was dealing with streaming TLV responses, so the problem is most likely in TLV stream serialization. On the payloads you've provided I'm seeing of by one error in data, but I can't find the problem in the code yet.

nonameplum, 2021-01-04:
@maximkulkin Aha. Yup, as I found that the last commit is problematic for sure this is related to the TLV. I just added a log for the `select(server->max_fd + 1, &read_fds, NULL, NULL, &timeout);` returned value and compared to the working version and I saw that is returns `0`.

I can put more logs if you like if that could help you spot the issue.

rottenhowler, 2021-01-04:
Ok, I have reproduced and fixed the problem. It boiled down to lwip_writev() not working as expected, so I replaced it with regular write(). Should work now.

maximkulkin, 2021-01-04:
Should be fixed now

closed 2021-01-04 by maximkulkin

nonameplum, 2021-01-06:
Thanks @maximkulkin
Yes, it's fixed now beside a weird issue that I have which is when I run my UDP server (https://github.com/nonameplum/esp-homekit-demo/blob/dca2f6811c60333fe8247247d9795199971a8f3b/helpers/udp_command.c#L73) that I use to send messages to IoT device over UDP. 
When UDP server is running, pairing faze 2/3 fails with -173 `BAD_FUNC_ARG` error, because in `wc_SrpSetPassword` `srp->side != SRP_CLIENT_SIDE`.
In pre TLV implementation I don't encounter this issue.

maximkulkin, 2021-01-08:
I bet you also don't encounter this issue if you do not run your UDP server.

nonameplum, 2021-01-08:
@maximkulkin Exactly when I do not run the UDP server then it works fine.

The thing is that if I check out some older commit (it's seems like its enough if I go back to https://github.com/maximkulkin/esp-homekit/commit/cd6c0d3b695e95fbb3fce3c95462db148ab81cc3 or for sure anyone before the new mDNS implementations)
it works also fine when the UDP server is running.

Also, I have checked for the last days the newest master (mDNS that I was curious), and what I observed is that now esp-homekit takes a bit more memory as I had to disable some of my features because devices couldn't pair as the pairing most of the time failed because of lack of memory during the process.
Another one is that mDNS and so device availability on the Home app in my particular case works much worse.
It's like in the 10 minutes range the device is available 2-3 times and only let's say for 5~60 seconds.

So for now I have switched back to https://github.com/maximkulkin/esphomekit/commit/a4b8d3e8020c7af92d49ed7f2f35c461f1917d01 as with it, I can enable all of my features and it still has enough memory to pair devices without a problem. Also, the UDP server can work seamlessly. To improve my issue of device availability I just change the `MDNS_TTL=10`, with such TTL my devices are available more than 90 percent of the time.