maximkulkin/esp-homekit #187: CURL - debug

awasspl, 2021-02-01:
Hi guys,

Hi can i check accessories list with curl command?
Somethink like: 

` curl -X GET http://192.168.0.111:5556/accessories --header "Content-Type:Application/json" --header "authentication: "111-11-111"" -s`

i make some mistake i think and still getting accessory "not compatible" :(

maximkulkin, 2021-02-02:
Ahaha, no no no no. That does not work like that. For debugging what went wrong, the easiest approach is to enable debug logging, the capturing json response from there. Response usually scattered across several HTTP chunks, so you'll have to do a bit of copy-pasting and concatenation.

awasspl, 2021-02-02:
it's a pity :( was be easier to just check services and accessories hierarchie.
I remember it was posible with non secure connection. now i don't know how to send auth header.

closed 2021-02-03 by awasspl