maximkulkin/esp-homekit #188: Is it possible to update the state of an accessory remotely?

ZweiEuro, 2021-02-19:
E.g.
My esp32 has a button that also controls the led, can I instruct the server to send out a "the button is now off/on" information to the apple homekit devices? how can i set such a state ?

maximkulkin, 2021-02-19:
Not sure what you mean by "button on" or "button off". Unless it is a physical latch button (that stays pressed once you press it) buttons have no state. And certainly you can't change physically latched button state from code.
What you can do is to gave an LED that changes state every time button triggers. See sonoff_basic example for implementation details.

ZweiEuro, 2021-02-19:
Ok i asked it the wrong way.
Assume the esp32 is connected to an LED , and a button, the button can switch the eld on and off just by shortly pushing it.
the same LED is controllable via homekit.

So now i have 2 ways of switching the LED on and off, button and homekit.
If the homekit does it, its fine since the state in homekit changes. but if i do it via the button, homekit will still think that the button is in the previous state. how can i update this state for homekit ?

ZweiEuro, 2021-02-19:
found my answer here
https://github.com/maximkulkin/esp-homekit/issues/157
sry for wasting your time ^^
awesome lib

closed 2021-02-19 by ZweiEuro