maximkulkin/esp-homekit #189: Graceful shutdown ?

ZweiEuro, 2021-02-19:
Dynamic services:
https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/esp32/dynamic_services/main/main.c

Don't really show anything like:
Gracefully shutdown server, 
change accessories (e.g. add something, or remove something),
Start the server back up.

resetting the server is a bit brutal as the pairing needs to be done. 
Is it possible to stop the server gracefully, change config and start it back up ?

maximkulkin, 2021-02-19:
What people usually do is storing updated config and rebooting the whole accessory. On boot it reads stored configuration and configures all new services.

ZweiEuro, 2021-02-19:
I wanted to avoid just shutting down, but if that is the only way to properly restart the server then i will do that.
I just wanted to make sure there was no other option

closed 2021-02-21 by ZweiEuro

ZweiEuro, 2021-12-29:
@maximkulkin  What would be the necessary steps to archive a 'graceful' shutdown ? Would throwing the server out of its main loop and letting it deallocate everything before starting it back up work ?