maximkulkin/esp-homekit #190: Conflict with arduino bluetooth library

ZweiEuro, 2021-02-21:
When i use both bluetooth from here:
https://github.com/nkolban/ESP32_BLE_Arduino
and your homekit server i get these types of errors:

````
E (95132) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (95132) task_wdt:  - IDLE0 (CPU 0)
E (95132) task_wdt: Tasks currently running:
E (95132) task_wdt: CPU 0: btController
E (95132) task_wdt: CPU 1: IDLE1
E (95132) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4013B303:0x3FFBE790 0x40082FF9:0x3FFBE7B0 0x40188A40:0x3FFB5E90 0x40188FF9:0x3FFB5EB0 0x4008FED5:0x3FFB5EE0 

E (95132) task_wdt: Print CPU 1 backtrace

Backtrace:0x40081DC3:0x3FFBED90 0x40082FF9:0x3FFBEDB0 0x4000BFED:0x3FFBC870 0x40090135:0x3FFBC880 0x4013B5A7:0x3FFBC8A0 0x4013B5B3:0x3FFBC8D0 0x4013B669:0x3FFBC8F0 0x40090B0D:0x3FFBC910 0x4008FED5:0x3FFBC930 

E (100132) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (100132) task_wdt:  - IDLE0 (CPU 0)
E (100132) task_wdt: Tasks currently running:
E (100132) task_wdt: CPU 0: btController
E (100132) task_wdt: CPU 1: IDLE1
E (100132) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4013B303:0x3FFBE790 0x40082FF9:0x3FFBE7B0 0x40188A40:0x3FFB5E90 0x40188FF9:0x3FFB5EB0 0x4008FED5:0x3FFB5EE0 

E (100132) task_wdt: Print CPU 1 backtrace

Backtrace:0x40081DC3:0x3FFBED90 0x40082FF9:0x3FFBEDB0 0x4000BFED:0x3FFBC870 0x40090135:0x3FFBC880 0x4013B5A7:0x3FFBC8A0 0x4013B5B3:0x3FFBC8D0 0x4013B669:0x3FFBC8F0 0x40090B0D:0x3FFBC910 0x4008FED5:0x3FFBC930
``` `

I know that it is not your problem, or of your library, but I wanted to know your best guess for the conflict, the only thing i could come up with is that maybe you are using the same nvs and overwriting each other.
I don't know how to get te hex decoded either sadly

maximkulkin, 2021-02-21:
Arduino libraries are not intended to work inside RTOS.

ZweiEuro, 2021-02-21:
In general? Oh, i didn't know that.
Is there an alternative lib i can use for bluetooth? The esp32 raw gatts implementation is incredibly hard to use.

ZweiEuro, 2021-02-21:
@maximkulkin  the same thing happens when using the normal gatts demo from esp-idf :)

```
E (22678) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (22678) task_wdt:  - IDLE0 (CPU 0)
E (22678) task_wdt: Tasks currently running:
E (22678) task_wdt: CPU 0: btController
E (22678) task_wdt: CPU 1: IDLE1
E (22678) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40138363:0x3FFBE8D0 0x40082E71:0x3FFBE8F0 0x401859E0:0x3FFB5E90 0x40185F95:0x3FFB5EB0 0x4008FED5:0x3FFB5EE0 
  #0  0x40138363:0x3 in task_wdt_isr at /home/zweieuro/.platformio/packages/framework-espidf/components/esp_common/src/task_wdt.c:187

E (22678) task_wdt: Print CPU 1 backtrace

Backtrace:0x40081DC3:0x3FFBEED0 0x40082E71:0x3FFBEEF0 0x4000BFED:0x3FFBC870 0x40090135:0x3FFBC880 0x40138607:0x3FFBC8A0 0x40138613:0x3FFBC8D0 0x401386C9:0x3FFBC8F0 0x40090B0D:0x3FFBC910 0x4008FED5:0x3FFBC930 

E (27678) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (27678) task_wdt:  - IDLE0 (CPU 0)
E (27678) task_wdt: Tasks currently running:
E (27678) task_wdt: CPU 0: btController
E (27678) task_wdt: CPU 1: IDLE1
E (27678) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40138363:0x3FFBE8D0 0x40082E71:0x3FFBE8F0 0x401859E0:0x3FFB5E90 0x40185F95:0x3FFB5EB0 0x4008FED5:0x3FFB5EE0 
  #0  0x40138363:0x3 in task_wdt_isr at /home/zweieuro/.platformio/packages/framework-espidf/components/esp_common/src/task_wdt.c:187

E (27678) task_wdt: Print CPU 1 backtrace

Backtrace:0x40081DC3:0x3FFBEED0 0x40082E71:0x3FFBEEF0 0x4000BFED:0x3FFBC870 0x40090135:0x3FFBC880 0x40138607:0x3FFBC8A0 0x40138613:0x3FFBC8D0 0x401386C9:0x3FFBC8F0 0x40090B0D:0x3FFBC910 0x4008FED5:0x3FFBC930 

E (32678) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (32678) task_wdt:  - IDLE0 (CPU 0)
E (32678) task_wdt: Tasks currently running:
E (32678) task_wdt: CPU 0: btController
E (32678) task_wdt: CPU 1: IDLE1
E (32678) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40138363:0x3FFBE8D0 0x40082E71:0x3FFBE8F0 0x401859E0:0x3FFB5E90 0x40185F95:0x3FFB5EB0 0x4008FED5:0x3FFB5EE0 
  #0  0x40138363:0x3 in task_wdt_isr at /home/zweieuro/.platformio/packages/framework-espidf/components/esp_common/src/task_wdt.c:187

E (32678) task_wdt: Print CPU 1 backtrace

Backtrace:0x40081DC3:0x3FFBEED0 0x40082E71:0x3FFBEEF0 0x4000BFED:0x3FFBC870 0x40090135:0x3FFBC880 0x40138607:0x3FFBC8A0 0x40138613:0x3FFBC8D0 0x401386C9:0x3FFBC8F0 0x40090B0D:0x3FFBC910 0x4008FED5:0x3FFBC930 
```

maximkulkin, 2021-02-21:
Looks like you're doing something long running inside callbacks not meant for that and it triggers task watchdog (it checks that tasks are not stuck).

ZweiEuro, 2021-02-21:
You mean in the bluetooth controller ?
Its the basic example they have for bte, I am not doing any computation in it

maximkulkin, 2021-02-21:
OK, I'm not a fortune teller, but if vanilla ESP-IDF example does not work, you should be asking Espressif.

ZweiEuro, 2021-02-21:
That's the issue, it seems to only not work when I try to use it while using your library as well

ZweiEuro, 2021-02-21:
I take it back, it's actually the example

closed 2021-02-21 by ZweiEuro