maximkulkin/esp-homekit #195: Custom Characteristic, compatible with EVE App

Camaendir, 2021-09-29:
I want to write a small LED-Strip Animation that is compatible with the EVE App. For that i need an additional Characteristic. How can i add one? I've tried the following:
```
#define HOMEKIT_CHARACTERISTIC_ANIMATION MYUUID
#define HOMEKIT_DECLARE_CHARACTERISTIC_ANIMATION(_value, ...) \
    .type = HOMEKIT_CHARACTERISTIC_ANIMATION, \
    .description = "ANIMATION", \
    .format = homekit_format_uint8, \
    .permissions = homekit_permissions_paired_read \
				 | homekit_permissions_paired_write \
                 | homekit_permissions_notify, \
    .min_value = (float[]) {0}, \
    .max_value = (float[]) {2}, \
    .min_step = (float[]) {1}, \
    .valid_values = { \
        .count = 3, \
        .values = (uint8_t[]) { 0, 1, 2 }, \
    }, \
    .value = HOMEKIT_UINT8_(_value), \
    ##__VA_ARGS__
```
added to the `characteristics.h` and using that just like the others. But the compilation errors with an "initializer element is not constant" error. Do you have an example with a custom characteristic.

Thanks for the help

maccoylton, 2021-09-29:
Here's a library I used in my accessories for custom characteristics:- 

https://github.com/maccoylton/esp-homekit-common-functions/tree/master/custom_characteristics

Camaendir, 2021-09-30:
@maccoylton Thanks for the Link. I still get an error "initializer element is not constant" Did you get such an error as well?

maccoylton, 2021-09-30:
The link I showed you works and is used in about a dozen  different accessories. Copy the pattern from one of my custom characteristics and it should work fine, your syntax doesn't look correct.

maximkulkin, 2021-10-02:
First, you do not need to put your custom characteristic type definitions into characteristics.h, it is sufficient to just put it somewhere and include it into code before you start creating characteristics of that type. E.g. you can put them right inside .c file, just make sure it is before your characteristics definitions.

Second, the error you're getting is not because of definition, but because of what you put inside that macro as a value. Can you post the code where you create characteristic?