maximkulkin/esp-homekit #207: Request pairing on demand by variable

FRANAIRBUS, 2022-08-16:
Hello, congratulations for the library.
I am using it without any problem and everything works correctly but I would like to be able to add or remove accessories on demand with a variable.

I explain:
I have a development board with 4 relays and my code works correctly even with 4 physical pushbuttons but I would like that if I have a development board with 2 relays I can tell my program to only show or print 2 relays with a simple global variable .

in other words to be able to choose if I want 1,2,3 or 4 relays
If I want the temperature option to use the thermostat, etc. without having to change the code, simply choosing a variable that indicates the number of accessories and the type to match.

something like that but this doesn't work.

```cpp
#if (ValueHK == 5)         

          
          HOMEKIT_ACCESSORY(.id=5, .category=homekit_accessory_category_switch, .services=(homekit_service_t*[]) {
          HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
              HOMEKIT_CHARACTERISTIC(NAME, "LH RELE 4"),  
              HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
              NULL
          }),
      HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
        &cha_switch_on4,
        &cha_name,
        NULL
      }),
          NULL
      }),

#endif
```

any idea how to get it?

Thank you

maximkulkin, 2022-08-16:
```c
#ifndef NUMBER_OF_RELAYS
#define NUMBER_OF_RELAYS 4
#endif

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_switch, .services=(homekit_service_t*[]){
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
            &name,
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "YOU"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "123456"),
            HOMEKIT_CHARACTERISTIC(MODEL, "Basic"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "0.1"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, identify),
            NULL
        }),
#if (NUMBER_OF_RELAYS == 1)
        HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Switch"),
            &switch1_on,
            NULL
        }),
#else
        HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Switch 1"),
            &switch1_on,
            NULL
        }),
#if (NUMBER_OF_RELAYS > 1)
        HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Switch 2"),
            &switch2_on,
            NULL
        }),
#if (NUMBER_OF_RELAYS > 2)
        HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Switch 3"),
            &switch3_on,
            NULL
        }),
#if (NUMBER_OF_RELAYS > 3)
        HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
            HOMEKIT_CHARACTERISTIC(NAME, "Switch 4"),
            &switch4_on,
            NULL
        }),
#endif
#endif
#endif
#endif
        NULL
    }),
    NULL
};
```

Then in Makefile add
```Makefile
NUMBER_OF_RELAYS ?= 1

EXTRA_CFLAGS += -DNUMBER_OF_RELAYS=$(NUMBER_OF_RELAYS)
```

Then you can compile it like this (plus or minus how you normally compile it)
```
make NUMBER_OF_RELAYS=3 all
```

FRANAIRBUS, 2022-08-16:
hello @maximkulkin 
Thank you very much for your quick response, I will implement it and it will be solved.

Thanks a lot
congratulations again

maximkulkin, 2022-08-23:
I assume it worked for you. If you still have issues with it, feel free to reopen this issue (or create new one if you have a different problem)

closed 2022-08-23 by maximkulkin