maximkulkin/esp-homekit #208: Failed to dump SPR public key (code -199)

AlexKryukoff, 2022-12-18:
Hello, if you want to connect to the monitor, I wrote an error -199. Can you suggest which way to look?

>>> HomeKit: [Client 1] Pair Setup Step 1/3
!!! HomeKit: [Client 1] Failed to dump SPR public key (code -199)
>>> HomeKit: [Client 1] Closing client connection from 192.168.1.108

mvso, 2023-01-21:
Same problem here.

Hellangel1987, 2023-02-07:
facing now same issue

[image]

@maximkulkin can you assist? we did everything correct. Code builds and upload ok. The ESP-cam start up fine and Home app finds the unit when I try to add it to my home. However, before I can type the 8-digit code, the serial prints out "Failed to dump SPR public key (code -199)" like shown above. The Home app process fail with "unknown error".