maximkulkin/esp-homekit #214: Reason using PORT 5556

anuarrozman, 2023-06-21:
Hii, I would like to ask about PORT you are using. Is there any specific reason you are using PORT = 5556 ?

This is located inside esp-homekit/src/server.c

`#define PORT 5556`

HomeACcessoryKid, 2023-06-21:
I would say the correct port is 661 since the protocol is HAP and that name is registered as such on
https://www.iana.org/assignments/service-names-port-numbers/service-names-port-numbers.xhtml?&page=13 .
But anyhow, any number will work since that is the point of the mdns system.

anuarrozman, 2023-06-21:
Thank you for your reply. If there's two devices with same port number, is it gonna be a problem ? Like suddenly undiscoverable.

maximkulkin, 2023-07-10:
Two devices will have unique IP addresses, so it should not be a problem.