maximkulkin/esp-homekit #216: ESP8266 Accessory No Response

problemss, 2023-11-21:
I am not sure what the catalyst was but I noticed my ESP8266 based device has been saying 'No Response' in homekit and has been like that for a while I guess. The log files are below. I am not sure why it is saying no pairing found as this device has been working fine for years. Any ideas? Note: I am on the latest homekit on my appleTV, 192.168.1.171

`13:28:18.022 -> ESP-Open-SDK ver: 0.0.1 compiled @ Mar 12 2020 11:51:28
13:28:18.022 -> phy ver: 273, pp ver: 8.3
13:28:18.022 -> 
13:28:18.022 -> on: 0
13:28:18.022 -> on:4 off1:2 off2:1 off3:1
13:28:18.022 -> ⸮
13:28:18.022 -> mode : sta(a4:cf:12:da:41:41)
13:28:18.022 -> add if0
13:28:18.022 -> >>> HomeKit: Starting server
13:28:18.022 -> >>> HomeKit: Using existing accessory ID: 5F:77:AC:45:6C:FF
13:28:18.022 -> >>> HomeKit: Configuring mDNS
13:28:18.100 -> scandone
13:28:19.191 -> mDNS announcement: Name=Fresh Air Intake md=SingleRelaySwitchpv=1.0id=5F:77:AC:45:6C:FFc#=1s#=1ff=0sf=1ci=8 Port=5556 TTL=4500
13:28:19.323 -> add 0
13:28:19.323 -> aid 20
13:28:19.323 -> cnt 
13:28:19.417 -> 
13:28:19.417 -> connected with WiFi_SSID, channel 1
13:28:19.463 -> dhcp client start...
13:28:20.214 -> ip:192.168.1.172,mask:255.255.255.0,gw:192.168.1.1
13:28:21.381 -> >>> HomeKit: Got new client connection: 4 from 192.168.1.171
13:28:21.381 -> >>> HomeKit: [Client 4] Pair Verify Step 1/2
13:28:21.850 -> >>> HomeKit: [Client 4] Pair Verify Step 2/2
13:28:21.896 -> !!! HomeKit: [Client 4] No pairing for CFA87CBF-B84B-4626-B10E-F252BF059FC5 found
13:28:21.990 -> >>> HomeKit: [Client 4] Closing client connection
13:28:22.037 -> >>> HomeKit: Got new client connection: 4 from 192.168.1.171
13:28:22.084 -> >>> HomeKit: [Client 4] Pair Verify Step 1/2
13:28:22.414 -> >>> HomeKit: [Client 4] Pair Verify Step 2/2
13:28:22.414 -> !!! HomeKit: [Client 4] No pairing for 288CE796-6F6B-4DA8-896B-E9088ED120CB found
13:28:22.555 -> >>> HomeKit: [Client 4] Closing client connection

maximkulkin, 2024-01-11:
Based on just this log it is hard to tell what happened, but it seems that pairing information is somehow corrupt. I see that the server thinks it is not paired ("sf=1" flag in mDNS announcement). Try re-pairing it.

problemss, 2024-01-11:
Thank you! I’ll give it a try in a couple weeks and reopen if needed.

closed 2024-01-11 by maximkulkin