maximkulkin/esp-homekit #217: I can't name the switch with different name in same accessory

xd94402002, 2024-05-07:
This is part of my accessory code

homekit_characteristic_t cha_auto_change_on = HOMEKIT_CHARACTERISTIC_(ON, true);
homekit_characteristic_t cha_auto_change_on_name = HOMEKIT_CHARACTERISTIC_(NAME, "Auto change");

homekit_characteristic_t all_swing = HOMEKIT_CHARACTERISTIC_(ON, true);
homekit_characteristic_t all_swing_name = HOMEKIT_CHARACTERISTIC_(NAME, "SWING");

homekit_characteristic_t function_name = HOMEKIT_CHARACTERISTIC_(NAME, "Function");

//=============================

  HOMEKIT_ACCESSORY(.id=8, .category=homekit_accessory_category_switch, .services=(homekit_service_t*[]) {
    	HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {	
        HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify), 
        &function_name, 
        NULL 
        }),
    	HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]) { 
        &cha_auto_change_on, 
        &cha_auto_change_on_name, 
        NULL
        }),
      HOMEKIT_SERVICE(SWITCH, .primary=false, .characteristics=(homekit_characteristic_t*[]) { 
        &all_swing, 
        &all_swing_name, 
        NULL 
        }),
		NULL
	}),

[image]

I always get the same name and I can't figure why. 
sometimes it will show as recommend in gray word  if I delete the default name.

maximkulkin, 2024-05-08:
It looks like it should work, but I have a question: is this code - a simplification or it is an original code as is? Because I have noticed that ACCESSORY_INFORMATION service is missing some required characteristics (e.g. MANUFACTURER, SERIAL_NUMBER, MODEL, FIRMWARE_REVISION).

Take a look at [this example](https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/dynamic_services), it defines a number of same type services with different names.

xd94402002, 2024-05-08:
It is a little part of the code, there are bunch of various accessory in my code.

This is my all code

 
#include "homekit/types.h"
#include <homekit/homekit.h>
#include <homekit/characteristics.h>

void my_accessory_identify(homekit_value_t _value) {
	printf("accessory identify\n");
}

homekit_characteristic_t fan_name = HOMEKIT_CHARACTERISTIC_(NAME, "FAN");

homekit_characteristic_t cold_fan_active = HOMEKIT_CHARACTERISTIC_(ACTIVE, 0);
homekit_characteristic_t cold_fan_rotation_speed = HOMEKIT_CHARACTERISTIC_(ROTATION_SPEED, 100);
homekit_characteristic_t cold_fan_swing_mode = HOMEKIT_CHARACTERISTIC_(SWING_MODE, 1);
homekit_characteristic_t cold_fan_name = HOMEKIT_CHARACTERISTIC_(NAME, "COLD FAN");

homekit_characteristic_t hot_fan_active = HOMEKIT_CHARACTERISTIC_(ACTIVE, 0);
homekit_characteristic_t hot_fan_rotation_speed = HOMEKIT_CHARACTERISTIC_(ROTATION_SPEED, 100);
homekit_characteristic_t hot_fan_swing_mode = HOMEKIT_CHARACTERISTIC_(SWING_MODE, 1);
homekit_characteristic_t hot_fan_name = HOMEKIT_CHARACTERISTIC_(NAME, "HOT FAN");

homekit_characteristic_t change_fan_active = HOMEKIT_CHARACTERISTIC_(ACTIVE, 0);
homekit_characteristic_t change_fan_rotation_speed = HOMEKIT_CHARACTERISTIC_(ROTATION_SPEED, 100);
homekit_characteristic_t change_fan_name = HOMEKIT_CHARACTERISTIC_(NAME, "CHANGE FAN");

homekit_characteristic_t dehumidity_name = HOMEKIT_CHARACTERISTIC_(NAME, "DEHUMIDIFIER");

homekit_characteristic_t cold_dehumidity_active = HOMEKIT_CHARACTERISTIC_(ACTIVE, 0);
homekit_characteristic_t cold_dehumidity = HOMEKIT_CHARACTERISTIC_(CURRENT_RELATIVE_HUMIDITY, 0);
homekit_characteristic_t cold_dehumidity_current = HOMEKIT_CHARACTERISTIC_(CURRENT_HUMIDIFIER_DEHUMIDIFIER_STATE, 0, .valid_values = {.count=1, .values = (uint8_t[]) {0}});  // 0, 1, 2, 3
homekit_characteristic_t cold_dehumidity_target = HOMEKIT_CHARACTERISTIC_(TARGET_HUMIDIFIER_DEHUMIDIFIER_STATE, 2, .valid_values={.count=1, .values = (uint8_t[]) {2}}); // 0, 1, 2);
homekit_characteristic_t cold_dehumidity_relative = HOMEKIT_CHARACTERISTIC_(RELATIVE_HUMIDITY_DEHUMIDIFIER_THRESHOLD, 40);
homekit_characteristic_t cold_dehumidity_swing_mode = HOMEKIT_CHARACTERISTIC_(SWING_MODE, 1);
homekit_characteristic_t cold_dehumidity_name = HOMEKIT_CHARACTERISTIC_(NAME, "COLD DRY");

homekit_characteristic_t hot_dehumidity_active = HOMEKIT_CHARACTERISTIC_(ACTIVE, 0);
homekit_characteristic_t hot_dehumidity = HOMEKIT_CHARACTERISTIC_(CURRENT_RELATIVE_HUMIDITY, 0);
homekit_characteristic_t hot_dehumidity_current = HOMEKIT_CHARACTERISTIC_(CURRENT_HUMIDIFIER_DEHUMIDIFIER_STATE, 0, .valid_values = {.count=1, .values = (uint8_t[]) {0}});  // 0, 1, 2, 3
homekit_characteristic_t hot_dehumidity_target = HOMEKIT_CHARACTERISTIC_(TARGET_HUMIDIFIER_DEHUMIDIFIER_STATE, 2, .valid_values={.count=1, .values = (uint8_t[]) {2}}); // 0, 1, 2);
homekit_characteristic_t hot_dehumidity_relative = HOMEKIT_CHARACTERISTIC_(RELATIVE_HUMIDITY_DEHUMIDIFIER_THRESHOLD, 40);
homekit_characteristic_t hot_dehumidity_swing_mode = HOMEKIT_CHARACTERISTIC_(SWING_MODE, 1);
homekit_characteristic_t hot_dehumidity_name = HOMEKIT_CHARACTERISTIC_(NAME, "HOT DRY");

homekit_characteristic_t function_name = HOMEKIT_CHARACTERISTIC_(NAME, "Function");

// format: float; min 0.0001, max 100000
homekit_characteristic_t cha_light_sensor = HOMEKIT_CHARACTERISTIC_(CURRENT_AMBIENT_LIGHT_LEVEL, 1);

homekit_characteristic_t cha_auto_light_on = HOMEKIT_CHARACTERISTIC_(ON, true);

homekit_characteristic_t cha_auto_change_on = HOMEKIT_CHARACTERISTIC_(ON, true);
homekit_characteristic_t cha_auto_change_on_name = HOMEKIT_CHARACTERISTIC_(NAME, "Auto change");

homekit_characteristic_t all_swing = HOMEKIT_CHARACTERISTIC_(ON, true);
homekit_characteristic_t all_swing_name = HOMEKIT_CHARACTERISTIC_(NAME, "SWING");

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_bridge, .services=(homekit_service_t*[]) {
    	// HAP section 8.17:
    	// For a bridge accessory, only the primary HAP accessory object must contain this(INFORMATION) service.
    	// But in my test,
    	// the bridged accessories must contain an INFORMATION service,
    	// otherwise the HomeKit will reject to pair.
    	HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "DELTA 400"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "delta test"),
[... 118 lines cut ...]
        HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
        HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify), 
        NULL 
        }),
    	  HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]) { 
        //HOMEKIT_CHARACTERISTIC(NAME, "Auto change"),
        &cha_auto_change_on_name,
        &cha_auto_change_on, 
        NULL
        }),
        HOMEKIT_SERVICE(SWITCH,. primary=false, .characteristics=(homekit_characteristic_t*[]) { 
        //HOMEKIT_CHARACTERISTIC(NAME, "SWING"),
        &all_swing_name,
        &all_swing, 
        NULL 
      }),
		NULL
	}),
  NULL
};

homekit_server_config_t config = {
		.accessories = accessories,
		.password = "111-11-111"
};